m-labs/artiq #1199: Multiple controllers with the same driver does not work from kernels

hartytp, 2018-11-28:
# Bug Report

## One-Line Summary

If the device db has two controllers, both using the same driver, they cannot be accessed from the same kernel.

## Issue Details

### Steps to Reproduce

Simple experiment to reproduce 

```python
from artiq.experiment import *

class RPCBugTestCase(EnvExperiment):

    def build(self):
        self.setattr_device("core")
        self.setattr_device("piezo_cavity_850")
        self.setattr_device("piezo_cavity_866")

    def prepare(self):
        pass

    @kernel
    def run(self):
        self.piezo_cavity_866.set_output_volts(49.)
        self.piezo_cavity_850.set_output_volts(50.)
```

With the corresponding device_db entries
```python
    "piezo_cavity_850": {
        "type": "controller",
        "host": master_addr,
        "port": next(ports),
        "command": "aqctl_thorlabs_tcube"
        " --port {port} --bind {bind} --product tpz001 --device \"hwgrep://0403:faf0 SER=29502239\""
    },
    "piezo_cavity_866": {
        "type": "controller",
        "host": master_addr,
        "port": next(ports),
        "command": "aqctl_thorlabs_tcube"
        " --port {port} --bind {bind} --product tpz001 --device \"hwgrep://0403:faf0 SER=29502286\""
    },
```

### Expected Behavior

Should work fine with no errors.

### Actual (undesired) Behavior

A type unification error:
```
root:While compiling /home/artiq/scratch/comet_repository/scratch/scratch_jw/RPC_bug_test_case.py

<synthesized>:1:1-1:59: error: host object has an attribute 'set_output_volts' of type [rpc3 #](...)->NoneType, which is different from previously inferred type [rpc2 #](...)->NoneType for the same attribute

`<artiq.protocols.pc_rpc.Client object at 0x7f15967a2a90>`

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

/home/artiq/scratch/comet_repository/scratch/scratch_jw/RPC_bug_test_case.py:14:1: note: expanded from here

def run(self):

^           
```

Calling `set_output_volts` for either of these devices individually works. Calling both from an RPC works, calling both directly from a kernel breaks.

### Your System

Artiq f5befba5c973808ac5150ec062eb9d9be7f2fd36

jordens, 2018-11-28:
This also happens (IIRC) if you want to use one Urukul with `sync_device` specified and another without or if you have a `AD9910` channel with `sw_device` and another without.

sbourdeauducq, 2018-11-28:
RPCs should be special cased and this should not happen here, the compiler should not know about the type of the controller client class. @whitequark

whitequark, 2018-11-28:
> RPCs should be special cased and this should not happen here, the compiler should not know about the type of the controller client class.

Should it really not? If we do that, other code will become impossible to write, e.g.:
```
if foo:
  ctrl = self.piezo_cavity_866
else:
  ctrl = self.piezo_cavity_850
ctrl.set_output_volts(50.)
```
What should happen here I think is that the RPC number should not be encoded into the type, but into the actual value (just stashed into the function pointer). So, RPC function types would only be unified on their return types, not return types and RPC numbers.

sbourdeauducq, 2022-03-06:
How common are those direct calls from kernels to RPC clients?
NAC3 does not support them at all as RPCs must be explicitly decorated with ``@rpc`` and type-annotated. So you would need to write wrapper functions, e.g.

```python
@nac3
class RPCBugTestCase(EnvExperiment):
    core: KernelInvariant[Core]
    # the RPC client objects are not annotated here, therefore are ignored by the compiler
    # and cannot cause any trouble.

    def build(self):
        self.setattr_device("core")
        self.setattr_device("piezo_cavity_850")
        self.setattr_device("piezo_cavity_866")

    @rpc
    def set_cavity_volts(self, obj: str, volts: float):
        getattr(self, obj).set_output_volts(volts)

    @kernel
    def run(self):
        self.set_cavity_volts("piezo_cavity_866", 49.)
        self.set_cavity_volts("piezo_cavity_850", 50.)
```

If this kind of boilerplate is too cumbersome, we can look into a semi-automatic way to generate it. If not, we can close this issue.

@dhslichter @lriesebos

lriesebos, 2022-03-07:
> How common are those direct calls from kernels to RPC clients?

For us, all calls to controllers (RPC clients) are wrapped in functions decorated with `@rpc`. In this way, we can be explicit about the fact that this actually is an RPC call and if it is async or not.

closed 2022-03-12 by sbourdeauducq