m-labs/artiq #1200: NXP firmware update via artiq_flash

jbqubit, 2018-11-29:
```artiq_flash``` support for flashing NXP microprocessors such as that used on Sayma AMC.

@sbourdeauducq Commented that this is likely to be an expensive and complex project. 

> Completely different protocol (on UART), which may or may not be documented and may need to be reverse-engineered, existing open implementation is dysfunctional, mapping UART/JTAG may be difficult on multiple platforms, flashing MMC does not fit in existing artiq_flash architecture, problems with packaging and conda, etc. etc. lots of tedious research to do and things to sort out.

sbourdeauducq, 2020-05-29:
@jbqubit Are you still interested in this?

closed 2020-12-14 by sbourdeauducq

jbqubit, 2020-12-14:
Not sure if this is a new development or not but looks like openocd advertises built-in support for the NXP LPC1776.  I'm looking at [Flash Driver: lpc2000](http://openocd.org/doc/html/Flash-Commands.html).

sbourdeauducq, 2020-12-15:
Yes, we used it before, it needs the internal JTAG connector.