m-labs/artiq #1203: [Deprecation]: Using importlib.load_module() in artiq.tools.file_import

drewrisinger, 2018-11-30:
# Bug Report

## One-Line Summary

`artiq.tools.file_import` uses deprecated `importlib.FileLoader` function [`load_module()`](https://github.com/python/cpython/blob/7212148c95947b0fdfcb0c8e37d4357287bdb4bd/Lib/importlib/_bootstrap_external.py#L947-L956).

## Issue Details

### Expected Behavior

ARTIQ should not rely on deprecated standard library functions. `importlib` suggests using [`exec_module()`](https://github.com/python/cpython/blob/7212148c95947b0fdfcb0c8e37d4357287bdb4bd/Lib/importlib/_bootstrap_external.py#L771-L777) instead.

### Actual (undesired) Behavior

Relevant code:
https://github.com/m-labs/artiq/blob/dce4f036db75e0f7380472bc1b0ec8bea977111b/artiq/tools.py#L86-L92

Further, this function call appears to be missing the parameter `fullname` in `Python` stdlib code.

Related `Python` stdlib code:
https://github.com/python/cpython/blob/7212148c95947b0fdfcb0c8e37d4357287bdb4bd/Lib/importlib/_bootstrap_external.py#L947-L956

Changing this will require some refactoring, because `exec_module()` doesn't have any return statement.

### Your System

* All versions of ARTIQ since at least 2.5

sbourdeauducq, 2018-12-01:
> Changing this will require some refactoring, because `exec_module()` doesn't have any return statement.

How do you locate the new module that you just loaded at all?

NB: the Python module API is changing often, and tracking it creates significant work (as you are beginning to notice, it is also not straightforward to use).

sbourdeauducq, 2018-12-01:
see also https://github.com/m-labs/artiq/issues/652

drewrisinger, 2018-12-03:
> > Changing this will require some refactoring, because `exec_module()` doesn't have any return statement.
> 
> How do you locate the new module that you just loaded at all?

Don't know what you mean by this or why you would want to locate the new module. I haven't looked into what [`load_module()`](https://github.com/python/cpython/blob/7212148c95947b0fdfcb0c8e37d4357287bdb4bd/Lib/importlib/_bootstrap_external.py#L947-L956) is supposed to do.

sbourdeauducq, 2018-12-03:
> why you would want to locate the new module

Because you have to return it.

drewrisinger, 2018-12-04:
Deprecated since Python 3.5: https://github.com/python/cpython/commit/32439d6eb63f1ea31aed1e45459f0f50f965ef51

PR #1605 (2021-02-10): tools/file_import: simplify, remove deprecated load_module() call

# ARTIQ Pull Request

## Description of Changes

This patch is a rewrite of ``artiq.tools.file_import`` with the following changes:

  - simplify the calculation of the loaded module's name. The new code is equivalent to the old one except if the filename has multiple extensions (e.g. ``exp.test.py``) in which case the resulting module name is not a valid Python identifier but the loader should still work as expected. This should not happen in practice since the filename also doesn't stick to Python module naming conventions. O [... cut]
closed 2021-02-10 by sbourdeauducq