m-labs/artiq #1205: SUServo: allow setting offset independently

ljstephenson, 2018-12-06:
# ARTIQ Feature Request

## Problem this request addresses

A clear and concise description of what the problem is.

## Describe the solution you'd like

A clear and concise description of what you want to happen.

## Additional context

Add any other context about the feature request here.

ljstephenson, 2018-12-06:
Oops - pressed enter early, please allow time for edit!

tballance, 2018-12-06:
This sounds like a simple feature but would help with usability. From my point of view it would be nice for experiment code to be able to seamlessly use Urukul or SUServo.

jordens, 2018-12-07:
Looks good. Do you want to prepare a PR?
I think it's just one memory access, not four.
And while at it this should probably also done for frequency and phase.