m-labs/artiq #1207: unification prevents simultaneous use of coredevices in different configurations

jordens, 2018-12-11:
# ARTIQ Feature Request

## Problem this request addresses

Using two Urukuls with different configurations (with/without `io_update_device` or with/without `sync_device`) leads to unification errors.

```python
from artiq.experiment import *

class A:
    def do(self):
        return 0

class B:
    def do(self):
        return 1

class C:
    def __init__(self, which):
        if which:
            self.sub = B()
        else:
            self.sub = A()

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.c0 = C(False)
        self.c1 = C(True)

    @kernel
    def run(self):
        return self.c0.sub.do() + self.c1.sub.do()
```

```
<synthesized>:1:1-1:64: error: host object has an attribute 'sub' of type <instance artiq_run_test_class_unification.B {
                __objectid__: numpy.int32
        }>, which is different from previously inferred type <instance artiq_run_test_class_unification.A {
                __objectid__: numpy.int32
        }> for the same attribute
`<artiq_run_test_class_unification.C object at 0x7f7f12f02390>`
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
test_class_unification.py:29:1: note: expanded from here
def run(self):
^
```

## Describe the solution you'd like

The generic pattern is that a feature (rf switch/sync/io_update supported either through `TTLOut` or through the CPLD register or not at all) exists or does not.

If there would be a way to force unification with "None", that could work around this.
@whitequark ?

## Additional context

c.f. #180 and #1199 for RPCs

jordens, 2018-12-11:
This works:

```python
from artiq.experiment import *

class A:
    def do(self):
        return 0

class B:
    def do(self):
        return 1

class C:
    def __init__(self, which):
        self.which = which
        if which:
            self.b = B()
        else:
            self.a = A()

    def do(self) -> TInt32:
        if self.which:
            return self.b.do()
        else:
            return self.a.do()

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.c0 = C(False)
        self.c1 = C(True)

    @kernel
    def run(self):
        print(self.c0.do() + self.c1.do())
```

jordens, 2018-12-11:
@whitequark is it by accident that this works and does the type annotation work around it?
Without the type annotation (which is AFAICT normally not needed in this case):
```
test_class_unification.py:37:15-37:27: error: cannot coerce NoneType to a numeric type
        print(self.c0.do() + self.c1.do())
              ^^^^^^^^^^^^
```

jordens, 2018-12-11:
If this is a correct solution, then the following should be adapted:

* Urukul: RegIOUPdate, DummySync
* ad53xx, ldac, clr

whitequark, 2018-12-11:
@jordens RPCs that return non-None values require a type annotation, by design.

jordens, 2018-12-11:
Oh. Yeah. That was meant to be a kernel.

whitequark, 2018-12-11:
@jordens Can you provide an MCVE? I've read the examples above a few times and I still don't understand the issue.

jordens, 2018-12-11:
@whitequark The MCVE is the first snippet posted. It's complete, verifiable and the most minimal I could come up with. Forget about the second snippet.
I am not claiming the current behavior is "wrong", just that I don't know how to resolve my issue demonstrated in the MCVE (which is equivalent to the one I am actually facing where you have two Urukuls, one with `sync_device` passed and another without).

jordens, 2018-12-12:
In words: Take two "devices" C (or Urukul) that implement a bit of functionality in different ways using either device A or B (`TTLOut` or something else). How can that be written?

cjbe, 2019-05-20:
@whitequark did you manage to reproduce this?
We are hitting this frequently at the moment (e.g. using SUServo and standalone Urukuls in the same experiment)

whitequark, 2019-05-21:
I can reproduce this. But there's no easy solution. That pattern is explicitly unsupported in the compiler and to add it would require major language changes in ARTIQ Python.

The best workaround is to have two derived classes, `C0` and `C1`, and override `__new__` to return an instance of each depending on the `which` argument.

sbourdeauducq, 2022-03-24:
Works with NAC3:
```python
from typing import TypeVar, Generic

from numpy import int32

from artiq.experiment import *
from artiq.coredevice.core import Core

@nac3
class A:
    @kernel
    def do(self) -> int32:
        return 14

@nac3
class B:
    @kernel
    def do(self) -> int32:
        return 30

T = TypeVar("T", A, B)

@nac3
class C(Generic[T]):
    sub: Kernel[T]

    def __init__(self, which):
        if which:
            self.sub = B()
        else:
            self.sub = A()

    @kernel
    def do(self) -> int32:
        return self.sub.do()

U = TypeVar("U", C[A], C[B])
V = TypeVar("V", C[A], C[B])

@nac3
class Types(Generic[U, V], EnvExperiment):
    core: KernelInvariant[Core]
    c0: Kernel[U]
    c1: Kernel[V]
    
    def build(self):
        self.setattr_device("core")
        self.c0 = C(False)
        self.c1 = C(True)
    
    @kernel
    def run(self):
        print_rpc(self.c0.do() + self.c1.do())
```

```
> artiq_run typedef.py
44
```

There is also an ``Option`` type coming soon, which is similar to the Rust ``Option`` which is easier to use and applicable to parts of the Zotino and Urukul drivers.

sbourdeauducq, 2022-03-24:
Later there will also be support for virtual classes, which provide more flexibility than the typevar annotations in the ``Types`` class.

closed 2022-03-24 by sbourdeauducq

ThorstenGroh, 2022-07-19:
Any news on that toptic? We would also like to control both our SuServo and standalone Urukul from the same kernel. Any recommendations for now how to achieve that with the default drivers? @cjbe @sbourdeauducq

bodokaiser, 2025-12-17:
> Any news on that toptic? We would also like to control both our SuServo and standalone Urukul from the same kernel. Any recommendations for now how to achieve that with the default drivers? [@cjbe](https://github.com/cjbe) [@sbourdeauducq](https://github.com/sbourdeauducq)

Any updates on this? One of our new labs is rolling out ARTIQ-10 (nac3) but cannot use SuServo and Urukuls

bodokaiser, 2025-12-19:
@sbourdeauducq Could you provide a hint?

atseml, 2025-12-24:
Proper method is to use `TypeVar`, but it won't work for now. Meanwhile I've started the discussion on this at https://forum.m-labs.hk/d/1088-suservo-with-nac3 (as guest, whoops).