m-labs/artiq #1214: Dashboard logging for crash debugging

cjbe, 2018-12-12:
# ARTIQ Feature Request

## Problem this request addresses

When running the dashboard on Windows one does not typically have access to the Python stdout if the process crashes (as one normally uses pythonw to avoid clutter from the Windows terminals). This means it can be difficult to capture error tracebacks from infrequent bugs.

## Describe the solution you'd like

The dashboard should log its output to a rolling set of log files, so one can always access the traceback from crashes.

jordens, 2018-12-13:
Logging to files is generally useful, also for master/ctlmgr.

sbourdeauducq, 2018-12-13:
For debugging the type of crash that @cjbe is talking about, and can potentially be e.g. a hard crash inside C++ Qt code, Python logging will not help.
Best is to redirect stdin and stdout to files in this case. @cjbe how exactly are you starting the dashboard?

The master already supports logging to files, and the controller manager already supports sending its logs to the master.

cjbe, 2018-12-13:
@sbourdeauducq typically we start the dashboard from a Windows batch script:
```
call activate artiq-env
start pythonw -m artiq.frontend.artiq_dashboard -s 10.255.6.191
```

sbourdeauducq, 2018-12-13:
Okay, what about simply redirecting to a log file in that script? The dashboard does not normally produce a lot of output.

``command >> log.txt 2>&1``

cjbe, 2018-12-13:
Yes - this seems to be the right kind of thing to do. However I don't see how to do this without causing a residual windows terminal left floating around (but this might just be my lack of Windows knowledge).

sbourdeauducq, 2018-12-13:
If this doesn't work, maybe try this instead:
https://www.reddit.com/r/Python/comments/4szp90/how_to_test_pythonwexe/