m-labs/artiq #1217: Core device should log crash dumps to flash

cjbe, 2018-12-12:
# ARTIQ Feature Request

## Problem this request addresses

If the core devices crashes without the serial port actively being monitored, it is impossible to record a traceback.

## Describe the solution you'd like

It would be useful if the core device stored the traceback to flash so it could be recovered by the master after rebooting the core device.

jordens, 2018-12-13:
Would be nice. I'd make this is opt-in. If the coredevice crashes, the cpu state may well be corrupted and venturing into (potentially corrupted) flash erase and write code might do a lot of harm.

sbourdeauducq, 2018-12-13:
If we add stack overflow protection, and considering that most things are in memory-safe Rust, it should be pretty hard to corrupt the comms CPU state in a way that the flash write code cannot run correctly. You would need things like certain rustc bugs, certain ARTIQ-Python compiler bugs, or hardware/gateware DRAM bugs.
Alternatively, we can reset the network stack and make the dump available for download (until the device is rebooted).