m-labs/artiq #1218: PYON Encode NamedTuple

drewrisinger, 2018-12-13:
# Bug Report

## One-Line Summary

PYON can't encode a NamedTuple. Fails silently.

## Issue Details

### Steps to Reproduce

```python
import artiq.protocols.pyon as pyon
from typing import NamedTuple
from collections import namedtuple

pyon.encode(NamedTuple("TypingTuple", [("val1", int)])(5))
pyon.encode(namedtuple("CollectionTuple", ["val1"])(5))
```

### Expected Behavior

PYON should be able to **encode a `namedtuple`** because it can be iterated/represented as a simple tuple. NamedTuples are useful for logically structuring data types that are passed between functions. Acceptable behavior: PYON encodes a NamedTuple as a regular tuple.

PYON should **also raise logical errors** through the PC RPC interface, so that debugging these issues is easier.

**Should be able to fix this by checking if the type is a subclass of some other type, instead of exact type matching.**

### Actual (undesired) Behavior

In my system, I make a RPC, which encodes a `NamedTuple` via PYON then sends it to a remote system.

```python
>>>device.perform_some_action()
WARNING:artiq.protocols.pc_rpc: connection failed while attempting RPC to ::1:3270[class 'artiq.protocols.pc_rpc.AutoTarget'>], re-establishing connection in the background
DEBUG:artiq.devices.SOME_DEVICE:SOME MESSAGE
RETURN_VALUE
>>> WARNING:artiq.protocols.pc_rpc:connection to ::1:3270[<class 'artiq.protocols.pc_rpc.AutoTarget'>] established in the background
```

**So 2 big problems
* RPC fails silently. Result of best-effort client, but doesn't raise error from PYON
* PYON doesn't encode namedtuples.**

### Your System

* ARTIQ version: ARTIQ 2 (though PYON code looks essentially identical in ARTIQ Dev)
* Hardware involved: running in simulation

drewrisinger, 2018-12-13:
Workaround for anyone struggling with this: cast it to a tuple: `tuple(named_tuple_object)`

sbourdeauducq, 2018-12-13:
> PYON should be able to **encode a `namedtuple`** because it can be iterated/represented as a simple tuple.

But it's not. PYON was never designed to support named tuples, and adding support is a feature, not fixing a bug.

drewrisinger, 2018-12-13:
It failing silently and needing to track it down is a bug

drewrisinger, 2018-12-13:
> > PYON should be able to **encode a `namedtuple`** because it can be iterated/represented as a simple tuple.
> 
> But it's not. PYON was never designed to support named tuples, and adding support is a feature, not fixing a bug.

Related: list of supported datatypes is out of date. https://m-labs.hk/artiq/manual-master/protocols_reference.html#module-artiq.protocols.pyon. Doesn't include OrderedDict.

sbourdeauducq, 2018-12-13:
> It failing silently

It does not:
```
>>> from artiq.protocols import pyon
>>> from collections import namedtuple
>>> x = namedtuple("x", "a b c")
>>> a = x(1, 2, 3)
>>> pyon.encode(a)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/nix/store/7avskr5yjm4sjjk5d6wvsjlk0m30wc6d-python3.6-artiq/lib/python3.6/site-packages/artiq/protocols/pyon.py", line 177, in encode
    return _Encoder(pretty).encode(x)
  File "/nix/store/7avskr5yjm4sjjk5d6wvsjlk0m30wc6d-python3.6-artiq/lib/python3.6/site-packages/artiq/protocols/pyon.py", line 170, in encode
    .format(x, type(x)))
TypeError: `x(a=1, b=2, c=3)` (<class '__main__.x'>) is not PYON serializable
```

sbourdeauducq, 2018-12-13:
Also, I don't see how they can be serialized in a way that is actually useful in practice. Turning them silently into regular tuples brings about new problems.

sbourdeauducq, 2018-12-17:
@drewrisinger I'm going to close this, if you need it please come up with a design and rationale for supporting named tuples in PYON.

closed 2018-12-17 by sbourdeauducq