m-labs/artiq #1219: RPC PYON Encoding fails silently

drewrisinger, 2018-12-13:
# Bug Report

## One-Line Summary

A failed PYON encoding when making an RPC call via `pc_rpc.AutoTarget` will fail silently (or show as disconnection). No easy way to trace bug back to PYON.

## Issue Details

### Steps to Reproduce

See #1218 for details. Warning copied below

```python
>>>device.perform_some_action()
WARNING:artiq.protocols.pc_rpc: connection failed while attempting RPC to ::1:3270[class 'artiq.protocols.pc_rpc.AutoTarget'>], re-establishing connection in the background
DEBUG:artiq.devices.SOME_DEVICE:SOME MESSAGE
RETURN_VALUE
>>> WARNING:artiq.protocols.pc_rpc:connection to ::1:3270[<class 'artiq.protocols.pc_rpc.AutoTarget'>] established in the background
```

Message makes it look like issue is with network connection, not ARTIQ.

sbourdeauducq, 2018-12-17:
@drewrisinger please test this patch:
```
diff --git a/artiq/protocols/pc_rpc.py b/artiq/protocols/pc_rpc.py
index adecf25d..fb922574 100644
--- a/artiq/protocols/pc_rpc.py
+++ b/artiq/protocols/pc_rpc.py
@@ -391,10 +391,11 @@ class BestEffortClient:
             self.__send(obj)
             obj = self.__recv()
         except:
-            logger.warning("connection failed while attempting "
-                           "RPC to %s:%d[%s], re-establishing connection "
+            logger.warning("failed RPC to %s:%d[%s], restarting connection "
                            "in the background",
-                           self.__host, self.__port, self.__target_name)
+                           self.__host, self.__port, self.__target_name,
+                           exc_info=True)
+            self.__socket.close()
             self.__start_conretry()
             return None
         else:
```
including all cases: attempting to send non-PYON-serializable data, receiving garbage data, connection breaking, first connection vs. connection that has been retried, ``close_rpc()`` called.

sbourdeauducq, 2019-01-02:
@drewrisinger ping