m-labs/artiq #1225: Docs: Compiling & rounding numpy floats

drewrisinger, 2019-01-03:
# Question

## Category: Compiler Docs

## Description

https://m-labs.hk/artiq/manual-release-4/compiler.html#pitfalls says that you should be able to run ``round(float(numpy.float64(1.0)))`` to get an ``int`` on both core & host device.

When in a program, this line fails:
```python
import numpy

from artiq.language.core import kernel
from artiq.language.environment import EnvExperiment

class TestFloat(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.core.reset()
        self.core.break_realtime()
        intval = round(float(numpy.float64(1.0)))
        print(intval)
```

```sh
$ artiq_run test_float.py
test_float.py: ... error: float cannot be invoked with the arguments (<instance numpy.float64 {
    __objectid__: numpy.int32
}>)
     intval = round(float(numpy.float64(1.0)))
                    ^^^^^

test_float.py: ... note: float can be invoked as float() -> float
[...]
test_float.py: ... note: float can be invoked as float(x:'a): -> float where 'a is numeric
[...]
```

What is the proper syntax? And please update docs to reflect.

NOTE: on further testing, looks like this runs fine with the ``@portable`` or ``@host_only`` decorators. But I think that means the code runs fine on the host, but isn't actually being compiled for/run on the core device.

sbourdeauducq, 2019-01-04:
@drewrisinger Please follow up on your past issues and PRs before opening new ones.

jordens, 2019-01-04:
Filing multiple clearly distinct issues in parallel is just fine IMO. There is no technical reason why that would be problematic. But please do follow up.

drewrisinger, 2019-01-09:
Followed up on all pending issues. Would appreciate an answer/improved documentation.

sbourdeauducq, 2019-01-10:
@drewrisinger Thank you.

Yes, ``@portable`` or ``@host_only`` applied to ``run()`` will execute the function in the regular Python interpreter on the host.

This looks like a compiler bug, @whitequark ?

whitequark, 2019-01-10:
It looks like `numpy.float64` is not currently recognized as a proper float. IIRC I did this for ints but I guess not for floats?

whitequark, 2019-01-10:
Yeah, can confirm. @sbourdeauducq Is it always OK to treat `numpy.float64` exactly like Python `float` or are there caveats?

sbourdeauducq, 2019-01-10:
AFAICT it's ok.

drewrisinger, 2019-01-10:
In meantime, core question has still not been answered. What's the correct syntax for right now, pending bug fix?