m-labs/artiq #1229: Kernel occasionally hangs indefinitely when reading from Urukul CPLD

tballance, 2019-01-07:
# Bug Report

## One-Line Summary

Kernel occasionally hangs indefinitely when reading from Urukul CPLD

## Issue Details

### Steps to Reproduce

```

class HangUrukul(EnvExperiment):
    """Hang urukul"""

   def build(self):
        self.setattr_device("core")
        self.setattr_device("dds_0")

    @kernel
    def run(self):
        self.core.break_realtime()
        delay(1*ms)

        self.dds_0.cpld.set_all_att_mu(0)
        self.dds_0.set(150*MHz)
        a = self.dds_0.cpld.get_att_mu()

        print(a)

```

### Expected Behavior

This experiment should either return after execution or cause an underflow error. 

### Actual (undesired) Behavior

* Without the dds.set line, the experiment executes correctly.
* With the dds.set line, the experiment hangs in the kernel indefinitely.

### Your System

* ARTIQ version: 5.0.dev+140.g01617c9f.dirty (forked from 5.0-dev on 02/12)
* Hardware involved: Kasli/Urukul

jordens, 2019-01-07:
I can reproduce this. Initializing the CPLD (`cpld.init()`) prevents it. I'd have to look at what part of `init()` is needed exactly.

jordens, 2019-01-09:
This is an `RTIOBusy` error triggered triggered by the io_update pulse nudging colliding with whatever the RTIO fine timestamp was before starting the experiment. The `cpld.init()` heals it by doing some nudging on its own.
It is arguably enabled and obfuscated by the problem that would be addressed by #1237.

commit 41a816fd (2019-01-09): ad9910: fix RTIO fine timestamp nudging

Previously the TSC was truncated to an even coarse RTIO periods before doing
the setting SPI xfer. Afterwards the the IO update pulse would introduce
at least one but less than two RTIO cycles. Ultimately the RTIO TSC was
truncated again to even. If the SPI xfer takes an odd number of RTIO
periods, then a subsequent xfer would collide.

close #1229