m-labs/artiq #1235: RTIOUnderflow when tuning sync_delay in Urukul

AUTProgram, 2019-01-09:
# Bug Report

## One-Line Summary

Method `tune_sync_delay` in class AD9910 throws underflow error, even when called with (presumably) enough slack.

## Issue Details

After initialising cpld and all AD9910 channels, I call method tune_sync_delay() on one of the channels, but an RTIOUnderflow exception is reported in `clear_smp_err` when it tries to write to SPI bus or pulse io_update high.

### Steps to Reproduce

```
class SyncDelay(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.cpld = self.get_device("trap_urukul_cpld")
        self.chs = []
        for i in range(4):
            self.chs.append(self.get_device("trap_urukul_ch"+str(i)))

    @kernel
    def run(self):
        self.core.reset()

        self.cpld.init()
        for ch_device in self.chs:
            ch_device.init()

        delay(1*ms)
        ch_device = self.chs[0]
        sync_delay, window = ch_device.tune_sync_delay()
        print(sync_delay, window)
```

### Expected Behavior

Check valid sync_delays and windows and return optimum values.

### Actual (undesired) Behavior

Sometimes finds valid delay and window and sometimes throws RTIOUnderflow exception.

```
root:Terminating with exception (RTIOUnderflow: RTIO underflow at 19629017498384 mu, channel 11, slack -1704 mu)
Core Device Traceback (most recent call last):
  File "/home/artiq/share/testing/bugs_mwe/sync_delay.py", line 24, in artiq_worker_sync_delay.SyncDelay.run(..., ...) (RA=+0xa80)
    sync_delay, window = ch_device.tune_sync_delay()
  File "<artiq>/coredevice/ad9910.py", line 645, in ... artiq.coredevice.ad9910.AD9910.tune_sync_delay<artiq.coredevice.ad9910.AD9910>(...) (RA=+0x1aec)
    self.clear_smp_err()
  File "<artiq>/coredevice/ad9910.py", line 595, in ... artiq.coredevice.ad9910.AD9910.clear_smp_err<artiq.coredevice.ad9910.AD9910>(...) (inlined)
    self.write32(_AD9910_REG_CFR2, 0x01010000)  # enable SMP_ERR
  File "<artiq>/coredevice/ad9910.py", line 183, in ... artiq.coredevice.ad9910.AD9910.write32<artiq.coredevice.ad9910.AD9910>(...) (inlined)
    self.bus.write(data)
  File "<artiq>/coredevice/spi2.py", line 225, in ... artiq.coredevice.spi2.SPIMaster.write<artiq.coredevice.spi2.SPIMaster>(...) (inlined)
    rtio_output((self.channel << 8) | SPI_DATA_ADDR, data)
  File "ksupport/rtio.rs", line 60, column 14, in (Rust function)
    <unknown>
                 ^
```

### Your System

* ARTIQ version: 5.0.dev+193.g0d3e7ba8
* Hardware involved: Kasli and Urukul v1.3

jordens, 2019-01-09:
I haven't seen this and can't reproduce it here. But this is likely fixed by the above change. Could you try?
Does this also affect 4.0? If yes, and this fixes it, it should be cherry-picked.

closed 2019-01-09 by jordens

jordens, 2019-01-09:
This is an `RTIOBusy` error triggered triggered by the io_update pulse nudging colliding with whatever the RTIO fine timestamp was before starting the experiment. The `cpld.init()` heals it by doing some nudging on its own.
It is arguably enabled and obfuscated by the problem that would be addressed by #1237.

AUTProgram, 2019-01-10:
Commit  b25ab1f fixed this issue, but I have not yet tried under the 4.0 version yet.

commit b25ab1fc (2019-01-09): ad9910: add more slack in tune_sync_delay

close #1235
commit 51239ebd (2019-01-09): ad9910: add more slack in tune_sync_delay

close #1235