m-labs/artiq #1237: transition "fine RTIO timestamp" into "RTIO data"

jordens, 2019-01-09:
# ARTIQ Feature Request

## Problem this request addresses

From the user perspective a lot of effort goes into making it look as if it was fully part of the RTIO timestamp and if the RTIO clock is 1 GHz. But in fact this isn't true:

* Currently the fine timestamp is only used in the high resolution SERDES TTLs.
* The RTIO TSC only counts coarse cycles. Same for DRTIO.
* The design limits the functionality of the SERDES TTLs to one edge per RTIO period. That leads to the weird constraint that a pulse shorter than one RTIO period is possible but only if it straddles the RTIO period edge. The 8x SERDES TTLs could support 8 edges per RTIO period and intriniscally do not have that other limitation.
* Fine RTIO events can't be simultaneous and are not ordered.
* Also there is quite some gatware required to do the edge encoding and decoding of the SERDES data.
* There is also gateware overhead in speacial treatment of the fine timestamp with regards to replacement, collision, and event re-ordering.
* The design makes the precise differentiation of the semantics and definitions of sequence errors, busy errors, collisions difficult and non-intuitive.
* For e.g. SAWG even though there is information faster than the RTIO period, this does not reach the user and leads to confusion.
* The implicit truncation of timestamps on all PHYs other than the SERDES TTLs leads to unintuitive behavior. E.g. SPI xfers always end on RTIO edges but subsequent TTL pulses use the full timestamp width, This can lead to jitter that is not apparent in the code and unexpected.
* There might easily be different RTIO multipliers soon that would complicate the issue. E.g. when some PHYs like SAWG go to 250 MHz intrinsic frequency or if there is 2 GS/s DAC data.

## Describe the solution you'd like

* Remove the fine timestamp bits and the notion of a fine RTIO clock, RTIO multiplier everywhere.
* Add wider "data" to the TTL RTIO PHY channels. 8 bits for the SERDES to indicate the pattern to drive.
  Since the TTLs have state and use "update semantics" the next SERDES RTIO cycle would continue with whatever the last bit of the RTIO event is (until and unless there is a new RTIO event).
* Add backwards compat user API that allows easy generation of "data" with a single edge.

## Additional context

* This is a breaking change.
* It would affect device_dbs and the meaning of the RTIO timestamp LSB.

Comments, opinions?

jbqubit, 2019-01-09:
Thanks for taking a step back to consider the broad implications of this early design choice.

For input do you likewise envision 8 bits per RTIO period that indicate the transition pattern? In this case backward compatible routines that return ns time-stamps on edge transition events and counts during interval are desirable.  

This would simplify the ARTIQ timing model from an end user's perspective. 👍 

Do these follow from what you propose? 
- mu duration reduced to 8 ns from 1 ns
- RTIO clock resolution reduced by 8 
- regarding a possible future high-resolution time-to-digital converter, timestamps might be recorded using bit field analogous to 8x SERDES TTLs

jordens, 2019-01-09:
As for TTL outputs, there would be methods for TTL input that process the pattern data (we would consider re-eabling FFL1 and using that). None of the other things follow:
* Machine unit duration is increased
* TSC resolution stays the same
* A high res TDC would probably not record pattern data but do the FFS/FLS processing in gateware plus maybe further compression. Pattern data doesn't scale. It would emit packed fractional timestamps as data.

jbqubit, 2019-01-09:
What are FFS, FLS, FFL1?

klickverbot, 2019-01-10:
Find first/last set, i.e. the operations that determine the index of the first/last '1' bit in a word. FFL1 are the respective OR1k instructions, which we disabled on our soft cores a while back to make the design less taxing.

sbourdeauducq, 2019-01-10:
Broadly agreed, some details:

> * There is quite some gateware required to do the edge encoding and decoding of the SERDES data.

Okay, but it reduces CPU overhead.

> * There is also gateware overhead in speacial treatment of the fine timestamp with regards to replacement, collision, and event re-ordering.

There is no resource overhead; the fine timestamp bits are simply excluded from those operations (with the caveats that you mention). It does make the gateware code more complex on the other hand.

>     * Remove the [...] notion of a fine RTIO clock, RTIO multiplier everywhere.

Those are still needed for clocking the SERDES. And on Ultrascale at least, they cannot be generated locally as they both have to come from the DRTIO transceiver.

sbourdeauducq, 2019-01-10:
NB: this proposal will disallow latency compensation (#40) with a resolution finer than RTIO cycles.

jordens, 2019-01-10:
@sbourdeauducq Updated the top post.
Yes. Gateware based latency compensation for TTLs would become as coarse as the RTIO clock. But that also was a pitfall for jitter and the pulses-shorter-than-a-RTIO-period. If fine delays are needed they would need to be implemented in software. The AD9910 driver does that already and needs software delays.
But AFAIK the latency compensation does not support the fine TSC bits at all:
https://github.com/m-labs/artiq/blob/master/artiq/gateware/rtio/sed/lane_distributor.py#L26
https://github.com/m-labs/artiq/blob/master/artiq/gateware/rtio/sed/lane_distributor.py#L144
I.e. this might be moot.

sbourdeauducq, 2019-01-10:
> But AFAIK the latency compensation does not support the fine TSC bits at all:
> https://github.com/m-labs/artiq/blob/master/artiq/gateware/rtio/sed/lane_distributor.py#L144
> I.e. this might be moot.

It can be modified to include the fine timestamp bits; I didn't do it to preemptively avoid FPGA timing issues. Only SAWG needs this internally, and only uses coarse timestamps. And properly supporting latency compensation needs a user API, documentation, and funding.

dhslichter, 2019-02-15:
I do generally agree that the current situation with "coarse" and "fine" timestamps is confusing and leads to some awkward situations.  This seems like a reasonable solution.  However, I am definitely nervous about the notion of pushing fine timestamp "generation" or "extraction", if you will, onto the CPU.  Having the ability to record and emit fine timestamps is a very nice (and for us, essential) feature of ARTIQ, but if the CPU overhead becomes too cumbersome it basically breaks the ability to do this.  We spend a lot of time worrying about performance improvements to get sustained RTIO throughput up, is this going to just slow all sustained RTIO throughput down by a factor of 2 (say), if fine timestamp resolution is required?  That would be a very bad outcome of all of this.  

I also worry about the ability to add offsets to timestamps which are finer than the RTIO clock.  By breaking out the 8 LSBs into the "data" portion, separate from the "RTIO timestamp", you would now have to add the "data" portions and carry overflows into the RTIO timestamp.  And while this addition of "data" portions is well-defined only if both of the "data" portions have only one edge; if one data portion has two edges, then you could end up shifting it so that the you now need two "data" portions in adjacent RTIO timestamps.  I believe that this may be tractable but it's certainly not pretty and also costs overhead.  

It's not clear to me how common the use case is for RTIO pulses shorter than the coarse RTIO period, or for multiple edges within a single coarse RTIO period.  Is this something you are really running up against?