m-labs/artiq #1241: SPI2: Proper value in Length Register

drewrisinger, 2019-01-11:
# Question

## Category: SPI

## Description

@jordens what is the proper value that should be in the SPI/length register when you run ``spi2.set_config(config_val, 32, clk_freq, chip_select)``? When I view the value in the register using ``artiq_coreanalyzer``, it shows `spi0/length[4:0]=0b11111` = `31`. Is this correct, or is the register not the proper length to hold 32? I can't tell if this is an off-by-one error or designed this way. If so, how does it handle 0-length? An error?

Note: this is not for urukul, but custom EURIQA hardware.

For reference, the other vals in the SPI control regs at the relevant time:
```
spi0/chip_select[7:0]=0x01
spi0/div[7:0] = 0x0B
spi0/flags[7:0] = 0x00
spi0/length[4:0] = 0x1F
spi0/read[31:0] = 0xXXXXXXXX
spi0/stb = 0
spi0/write[31:0] = 0x61ab 0000
```

jordens, 2019-01-11:
The register holds the number of bits minus one.
https://github.com/m-labs/misoc/blob/master/misoc/cores/spi2.py#L89-L90
https://github.com/cjbe/artiq/blob/master/artiq/coredevice/spi2.py#L170
It doesn't support 0-length:
https://github.com/cjbe/artiq/blob/master/artiq/coredevice/spi2.py#L165-L166

drewrisinger, 2019-01-11:
Thanks for the prompt response. I guessed as much, just wanted to confirm all discrepancies I saw in the traces.

closed 2019-01-11 by drewrisinger