m-labs/artiq #1250: idle_kernel doesn't run at power-on

jbqubit, 2019-01-16:
## One-Line Summary

Upon reset of board with ```idle_kernel``` defined the idle_kernel doesn't run until after first user-specified kernel runs (eg with ```artiq_run```). 

### Steps to Reproduce
Program flash storage with idle_kernel but no startup_kernel. Reset board. Observe that idle_kernel does not run. Call artiq_run to run another program and terminate it. Now idle_kernel runs automatically. 

### Expected Behavior
idle_kernel runs whenever kernel is idle including following boot. 

### Your System

* Operating System: Ubtuntu 14.04
* Conda version: 4.5.12
* ARTIQ version:4.0.dev0+1298.ge4d48a78
* Hardware involved: Kasli v 1.1

sbourdeauducq, 2019-01-17:
Board log?

sbourdeauducq, 2019-03-14:
@jbqubit ping

closed 2019-04-19 by sbourdeauducq