m-labs/artiq #1255: array network transfer speed

jordens, 2019-01-28:
# Bug Report

## One-Line Summary

Transferring integer arrays from coredevice to host is more than a factor of ten slowe than the raw binary data transfer rate.

## Issue Details

### Steps to Reproduce

```
ARTIQ_LOW_LATENCY=1 ARTIQ_ROOT=somewhere python -m unittest -v artiq.test.coredevice.test_performance.TransferTest.test_device_to_host_array_test
```

### Expected Behavior

Some data rate similar to the 2.4 MB/s of `test_device_to_host`

### Actual (undesired) Behavior

211 kB/s

```
test_device_to_host_array (artiq.test.coredevice.test_performance.TransferTest) ... Mismatch between gateware (5.0.dev+249.g84f7d006.dirty) and software (5.0.dev+260.gb692981c.dirty) versions
211608.44074349522 B/s
ok

----------------------------------------------------------------------
Ran 1 test in 1.017s

OK
```

### Your System

* ARTIQ version: b692981c
* Hardware involved: Kasli, Ethernet, Linux

whitequark, 2019-01-29:
This is likely because of the serialization code. I think the proper solution is to add fast paths for specific kinds of arrays.

airwoodix, 2019-02-01:
Is this something reasonably easy to fix?
This behaviour is very surprising to newcomers, since one expects that the compiler particularly optimizes this case away. Also, list/array transfers are used all over the place, including in the examples (e.g. `kasli_tester`). Bytearrays probably work around the problem but are not really convenient in high-level code.

airwoodix, 2019-02-12:
@jordens @whitequark Do you have an update/statement on this? Thanks!

jordens, 2019-02-12:
@airwoodix The speed is inherent in the current implementation. It doesn't seem many people are currently limited by it or would be able do do new things if this was tuned for high data rate array transfers.

But absolutely, IMO this is something definitely worth improving and it doesn't sound too invasive.

airwoodix, 2019-02-13:
Ok, thanks for the feedback. I'm a bit surprised but if no one else complained, then they're not limited, fair enough.

dnadlinger, 2019-02-17:
We typically stream out data as it is acquired to work around this, and don't currently have any large data (e.g. AWG waveforms) we'd want to dynamically load onto the kernel CPU. Networking on the core device used to be so amazingly bad that a certain amount of complacency has set in.

jordens, 2019-08-09:
This is also hit when pushing Sampler data to the host.

```python

from artiq.experiment import *
import numpy as np

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("sampler0")
        self.queue = []

    @rpc(flags={"async"})
    def cb(self, data):
        self.queue.append(data)

    def run(self):
        dt = self.sample()
        data = np.array(self.queue)
        print(np.std(data))
        print(data.shape)
        print(self.core.mu_to_seconds(dt)/data.size)
        print(data[-1][-3:])

    @kernel
    def sample(self):
        reps = 10
        channels = 8
        d = [[0]*channels for i in range(1024)]
        acqs = len(d)
        self.core.reset()
        dev = self.sampler0
        dev.init()
        dt = int64(0)
        for l in range(reps):
            for i in range(acqs):
                dev.sample_mu(d[i])
                delay(8*us)
            dt -= self.core.get_rtio_counter_mu()
            self.cb(d)
            dt += self.core.get_rtio_counter_mu()
            self.core.break_realtime()
        return dt
```

charlesbaynhamNPL, 2019-08-12:
If you only need a single channel from Sampler, you can get ~30 KSPS by reducing the amount of data sent by the core:

    self.cb([x[channel] for x in d])

in the above.

charlesbaynham, 2019-08-12:
If you only need a single channel from Sampler, you can get ~30 KSPS by reducing the amount of data sent by the core:

    self.cb([x[channel] for x in d])

in the above.

charlesbaynham, 2019-08-12:
@airwoodix Did you find out whether using `bytearray`s works around this?

pca006132, 2020-09-03:
PR #1510 and #1511 should resolve some of the performance issue. However the performance of device to host transfer of integer array is still limited by endian conversion for or1k. We currently loop over each integer, do endian conversion, and append it to the vector. This involves quite a lot of code, and the vector append may require additional allocation (O(1) is amortized only, the constant factor is not small, and we need a lot of reallocation when the vector starts small). So, the integer array/list device to host performance is pretty slow.

For zynq, we mitigated the issue by copy the array into a new vector, do the byte swap, and copy the entire vector into the writer. This requires allocation for copying the array, otherwise we have to swap, copy and swap it back which normally takes more time. Also, this requires fast memcpy performance, which I am not sure for or1k platform.

Wondering what if we just use the device endian, and do the endian conversion in the host. I think the host should be able to handle that pretty quickly.

dnadlinger, 2021-01-13:
NB: Most host systems will also be little-endian, so the endian conversion can just be a no-op.

PR #1589 (2021-01-21): firmware: optimize integer array/list rpc
# ARTIQ Pull Request

## Description of Changes
Optimize integer array/list device-to-host RPC.

As the device is in big endian, we can cast the slice into byte slice, directly copy from slice instead of calling `write_u32`/`write_u64` repeatedly. This improves the performance quite a lot.

| Test                 | Prev (MiB/s) | Now (MiB/s)  |  diff  |
| -------------------- | ------------ | ------------ | ------ |
| I32 Array (1MB) H2D  |         2.02 |         2.06 | +1.90% |
| I32 Array (1MB) D2H  |         0.48 |         2.23 |  +365% |
| I32 Arra [... cut]
closed 2021-01-21 by dnadlinger