m-labs/artiq #1256: SUServo negative numbers

hartytp, 2019-01-30:
There seems to be an issue with handling of negative numbers in SUServo.

Simple demo
```python
from artiq.experiment import *
from artiq.coredevice.suservo import COEFF_WIDTH

class Startup(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("aom_servo")
        self.setattr_device("aom_397_doppler")
        self.offset = -0.00817

    @kernel
    def run(self):
        self.core.reset()

        self.aom_servo.init()
        self.aom_servo.set_config(enable=1)

        dds = self.aom_servo.dds0

        ftw = dds.frequency_to_ftw(200*MHz)
        pow_ = dds.turns_to_pow(0.)
        # offs = int(round(self.offset*(1 << COEFF_WIDTH - 1)))
        offs = -1071
        self.aom_397_doppler.set_dds_mu(0, ftw, offs, pow_)

        data = [0] * 8

        self.aom_397_doppler.get_profile_mu(0, data)

        print(ftw >> 16, data[0])
        print("b1 ", data[1])
        print(pow_, data[2])
        print("adc | (delay << 8)", data[3])
        print(offs, data[4])
        print("a1 ", data[5])
        print(ftw & 0xffff, data[6])
        print("b0 ", data[7])

```

When I set offset to positive numbers, the data I read back matches the data I write:

```
2019-01-30 14:56:33,043 INFO:worker(168,servo_test.py):print:13107 13107
2019-01-30 14:56:33,046 INFO:worker(168,servo_test.py):print:b1  18513
2019-01-30 14:56:33,050 INFO:worker(168,servo_test.py):print:0 0
2019-01-30 14:56:33,053 INFO:worker(168,servo_test.py):print:adc | (delay << 8) 73777
2019-01-30 14:56:33,056 INFO:worker(168,servo_test.py):print:1000 1000
2019-01-30 14:56:33,060 INFO:worker(168,servo_test.py):print:a1  65553
2019-01-30 14:56:33,064 INFO:worker(168,servo_test.py):print:13107 0
2019-01-30 14:56:33,067 INFO:worker(168,servo_test.py):print:b0  0
```

if I now change offset to a negative number then the value I readback is unaltered:
```
2019-01-30 14:57:16,721 INFO:worker(169,servo_test.py):print:13107 13107
2019-01-30 14:57:16,724 INFO:worker(169,servo_test.py):print:b1  18513
2019-01-30 14:57:16,728 INFO:worker(169,servo_test.py):print:0 0
2019-01-30 14:57:16,731 INFO:worker(169,servo_test.py):print:adc | (delay << 8) 73777
2019-01-30 14:57:16,735 INFO:worker(169,servo_test.py):print:-500 1000
2019-01-30 14:57:16,738 INFO:worker(169,servo_test.py):print:a1  65553
2019-01-30 14:57:16,742 INFO:worker(169,servo_test.py):print:13107 0
2019-01-30 14:57:16,745 INFO:worker(169,servo_test.py):print:b0  0
```

Moreover, some negative offsets seem to have unexpected consequences such as disabling the servo.

I've also seen issues with the ADC readings from `get_adc` returns large positive numbers when the input voltage is negative. Any ideas?

hartytp, 2019-01-30:
Also, unless I'm missing something, there seems to be an issue with `data[6]` (the lower bits of the ftw) always being 0.

hartytp, 2019-01-30:
Note to self: the current delay in `get_profile_mu` isn't large enough and pretty deterministically gives underflows. Increasing to `6us` fixes.

hartytp, 2019-01-30:
I suspect the issue is related to https://github.com/m-labs/artiq/blob/450a035f9ef0dd9637325e04c6e708e97f7bfdb4/artiq/coredevice/suservo.py#L133

AFAICT value is a twos complement 32 bit integer. So for negative numbers all the higher bits are set, which corrupts the address (although I don't fully understand the SUServo memory structure so hard to be sure).

hartytp, 2019-01-30:
Okay, all these issues are related to not correctly masking off bits in the data written.

I'll submit a PR tomorrow.

hartytp, 2019-01-30:
@jordens am I missing something, or does this issue appear multiple times in the SUServo code. e.g. https://github.com/m-labs/artiq/blob/450a035f9ef0dd9637325e04c6e708e97f7bfdb4/artiq/coredevice/suservo.py#L286 should be `ftw & 0xffff` right? Otherwise it overflows the data part of the register.

sbourdeauducq, 2019-01-31:
Usual suspect: https://github.com/m-labs/artiq/commit/56315203a95bd2ebeb3ae9c64021fcdbfef5f11e

(see https://github.com/m-labs/artiq/issues/1252)

cc @whitequark

hartytp, 2019-01-31:
@jordens another thing I don't understand:

This fragment sets the a1 coefficient to 1 and then reads it back.

```python
        a1 = 1
        profile = 0
        base = (self.aom_397_doppler.servo_channel << 8) | (profile << 3)
        self.aom_servo.write(base + 5, a1)
        readback = self.aom_servo.read(base + 5)

        test = [0]*32
        for i in range(32):
            test[i] = 1 if ((readback << i) & (1<<31)) != 0 else 0
        print(test)
```

printed output is `print:[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 1, 0, 0, 0, 1]`

numbers I write to this coefficient all seem to behave as I expect apart from those two bits being set. Have I misunderstood how this is supposed to work, or is there a memory corruption issue here?

jordens, 2019-01-31:
Could you check whether this is a regression from release-4? There were the new rtio register API and the compiler changes.

jordens, 2019-01-31:
Well 4.0 if the compiler change already landed in releaee-4.

hartytp, 2019-01-31:
@jordens I'll have a look at that soon. However, looking over the coredevice code it seems to me that there are issues that need addressing anyway, even if they only became symptomatic after the compiler changes.

hartytp, 2019-01-31:
I've started pushing fixes here: https://github.com/hartytp/artiq/tree/suservo

hartytp, 2019-01-31:
@jordens one thing that got us here (but which isn't necessarily a bug) is that the adc machine units can be negative, but are stored as the LSB of a 32-bit number. So, doing arithmetic on the raw mus for negative voltages can get one into trouble. I suggest that we either (1) convert all numbers to proper twos complement 32 bit numbers; or (2) make this clear in the documentation. Preferences?

jordens, 2019-01-31:
Iirc sign extending the read port on that memory is fine. The coefficient memory might be trickier.

hartytp, 2019-01-31:
Okay, I'll sign extend at least `get_adc_mu`, and have a think about what to do with coefficients.

Now that I understand this bug, it's not really a problem. #1258 is much more concerning.

jordens, 2019-02-06:
Sign extend the memory port. It's cheaper than in software.

jordens, 2019-03-21:
@hartytp is this resolved now with the patches? does it affect release-4?

hartytp, 2019-03-21:
I haven't checked release 4 so can't comment.

The negative numbers issue isn't fixed yet -- I had a tested sw patch that you rejected in favour of a gw solution. I've written but not tested a gw solution. It's been second thing on my to do list for a while now...

closed 2019-08-15 by hartytp