m-labs/artiq #1260: Sayma RTM HMC7043/DDMTD tests

sbourdeauducq, 2019-02-01:
Instructions to rework hardware:
* Cut the Si5324 reset trace on the RTM. This is pin 1:
[image]
This pin contains an internal pull-up, so it does not need to be connected to anything. The new firmware will use the software reset instead (like on Kasli).
* Solder on 100 ohm LVDS termination resistors across R310/R313 and across R314/R315.
The 100 ohm resistor needs to added to the two rightmost pads here:
[image]
and to the two leftmost pads here:
[image]
* Connect uFL jumper cables between RTM "REF LO DIAG" and "CRD AUX CLKIN" (sic).

Those reworks do not interfere with any current board function and can be left on permanently.

After each power-on of the board, the RTM Si5324 needs to be put into bypass mode. The bitstream for doing this is built as follows:
* Hack-patch the satellite manager firmware as follows:
```diff
diff --git a/artiq/firmware/satman/main.rs b/artiq/firmware/satman/main.rs
index 180ccbcb..82e93bb4 100644
--- a/artiq/firmware/satman/main.rs
+++ b/artiq/firmware/satman/main.rs
@@ -417,6 +417,10 @@ pub extern fn main() -> i32 {
     serwb::wait_init();
 
     i2c::init();
+    si5324::bypass(si5324::Input::Ckin2).expect("cannot initialize Si5324");
+    info!("si5324 bypassed");
+    loop {}
+
     si5324::setup(&SI5324_SETTINGS, si5324::Input::Ckin1).expect("cannot initialize Si5324");
     #[cfg(has_hmc830_7043)]
     /* must be the first SPI init because of HMC830 SPI mode selection */
```
* Compile the target ``sayma_rtm_drtio.py``.
* Make the bitstream loadable. You will need [this tool](https://github.com/quartiq/xilinx-bitstream).
```shell
 python xilinx-bitstream/xilinx-bitstream.py artiq_sayma_rtm/gateware/top.bit rewrite.bin
```
* Revert the satman hack (so it doesn't get loaded into the AMC):
```
git checkout artiq/firmware/satman/main.rs
```

It needs to be loaded after each power-up, but with the reset rework, the Si5324 configuration is not lost when bitstreams are loaded, so this needs to be done just once if your board power supply [is stable](https://github.com/sinara-hw/sinara/issues/567) (tl;dr cranking up the µTCA fans to the maximum seems to be a workaround).
```shell
openocd -f sayma.cfg -c "pld load 0 rewrite.bin; exit"
```

contents of sayma.cfg:
```
interface ftdi
ftdi_device_desc "Quad RS232-HS"
ftdi_vid_pid 0x0403 0x6011
ftdi_channel 0
# EN_USB_JTAG on ADBUS7: out, high
# nTRST on ADBUS4: out, high, but R46 is DNP
ftdi_layout_init 0x0098 0x008b
reset_config none

adapter_khz 5000
transport select jtag

source [find cpld/xilinx-xc7.cfg]
set CHIP XCKU040
source [find cpld/xilinx-xcu.cfg]

init
```

@hartytp @marmeladapk Could you reproduce this on your boards and let me know what the DDMTD results are? Also after reverting https://github.com/m-labs/artiq/commit/ec230d65608f86c79d3fc3e8f6e75ac859fa45fb
Note that before https://github.com/m-labs/artiq/commit/ec230d65608f86c79d3fc3e8f6e75ac859fa45fb the firmware does not need any hardware rework.

sbourdeauducq, 2019-02-01:
Further instructions:
* build the Satellite variant of Sayma
* build the Master variant of Kasli. You need a Urukul 9910 as well and it should be clocked from the Kasli MMCX.
* connect second SFP of Kasli to first SFP of Sayma
* connect last Urukul channel to Sayma RTM input
* run the kasli_sawgmaster/sines_urukul_sayma kernel.

sbourdeauducq, 2019-03-08:
@hartytp Has anyone at Oxford reproduced this, or is this hampered by some random breakage?

dnadlinger, 2019-03-08:
cc @vmsch

hartytp, 2019-03-08:
Too much random breakage too little time

hartytp, 2019-03-08:
The rework on the Sayma we have seems to have been damaged during shipping and getting life out of them is proving non trivial

sbourdeauducq, 2019-11-14:
Obsolete with v2 hardware.

closed 2019-11-14 by sbourdeauducq