m-labs/artiq #1272: hdf5 results: merge "archive" and "datasets" groups

jordens, 2019-02-10:
# ARTIQ Feature Request

## Problem this request addresses

Currently in the HDF5 results, there are get() datasets in the "archive" group and set() datasets in the "datasets" group. This is a bit confusing and doesn't seem necessary since we assume/warn/enforce that get and set datsets share a namespace and an entry is never both. 

## Describe the solution you'd like

Merge the groups. The browser already builds a merged namespace of the two groups.

## Additional context

Maybe #1171

airwoodix, 2020-11-17:
I also find this very confusing.
@hartytp @jordens  Are there code bases that heavily rely on the `datasets`/`archive` split? If so, what's the use case?
Merging the two sets would also nicely simplify the code in `master.worker_db`.

hartytp, 2020-11-17:
@airwoodix @dpn is a better person to ask than me (I typically only interact with HDF5 results via tooling he wrote so am not really conscious of it's internal structure).

jordens, 2020-11-17:
I think it's only `browser` and maybe the experiment load-from-hdf that would look at it. As I said. I don't think it's necessary. It's a remnant from before #1171