m-labs/artiq #1274: Compiling kernel method fails when 'line' contains a carriage return.

philipkent, 2019-02-12:
# Bug Report

## One-Line Summary
Compiling a kernel method fails in the symbolize method when the 'line' variable contains a trailing carriage return.

## Issue Details
When we try to execute an experiment on the core device, we sometimes get a "ValueError: invalid literal for int() with base 10: '?\r'" on line 226 of targets.py (full error output is below).

For one of our experiments, there is a line that is parsed as "?\r".  i.e. set to "?" but with a carriage return appended at the end.  This might be due to some non-standard formatting in our code but I haven't tracked it down to a specific line.  This is fixed, however, if I change line 222 in artiq/compiler/targets.py from
` if line == "?": `
to 
`if line.strip() == "?": `

### Steps to Reproduce
We have not been able to find the exact source of the appended carriage return.

### Expected Behavior
kernel method should compile

### Actual (undesired) Behavior
A specific kernel method does not compile.  Here is the error log output: 
* Text description
* Log message, tracebacks, screen shots where relevant
`root:Terminating with exception (ValueError: invalid literal for int() with base 10: '?\r')
Traceback (most recent call last):
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6-2\lib\site-packages\artiq\master\worker_impl.py", line 273, in main
    exp_inst.run()
  File "C:\artiq\shared\artiq_ions\scanning\scan.py", line 367, in run
    self._run_scan_core(resume)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6-2\lib\site-packages\artiq\language\core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6-2\lib\site-packages\artiq\coredevice\core.py", line 133, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6-2\lib\site-packages\artiq\coredevice\comm_kernel.py", line 603, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6-2\lib\site-packages\artiq\coredevice\comm_kernel.py", line 584, in _serve_exception
    traceback = list(reversed(symbolizer(backtrace))) + \
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6-2\lib\site-packages\artiq\coredevice\core.py", line 111, in <lambda>
    lambda addresses: target.symbolize(library, addresses), \
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6-2\lib\site-packages\artiq\compiler\targets.py", line 226, in symbolize
    line = int(line)
ValueError: invalid literal for int() with base 10: '?\r'`

### Your System

* Operating System: Windows 7
* Conda version: 4.4.8
* ARTIQ version: 3.6 host software, 3.6+0.g394b66cd.dirty gateware

sbourdeauducq, 2019-02-12:
Can you attach a file that causes the problem?

philipkent, 2019-02-12:
This doesn't seem to be happening in Artiq version 3.7.  It is not, at least, for our experiment that was failing in Artiq 3.6.  After debugging under version 3.6, it seemed like the issue had something to do with using multiple return statements in a kernel method when using inheritance.  The incorrectly parsed line come from a kernel method that had three return statements.  I only wen't down the rabit hole so far, and wasn't able to pin point the exact problem.  But it seems like this isn't a problem in version 3.7.

closed 2019-02-12 by jordens