m-labs/artiq #1275: Compiler: unexpected behavior of numpy bool

pathfinder49, 2019-02-12:
# Bug Report

## One-Line Summary

Unexpected compiler behavior with numpy bool. This can produce silent errors.

## Issue Details

1. Implicit casting between Python and numpy bool hasn't been implemented in the compiler. E.g. `np.bool_(True) == True` gives a compiler error. In comparison, the corresponding situation for ints (`np.int_(0) == 0`) is handled fine. 
1a. calling `np.bool_(True)` from the kernel produces `AttributeError: 'TVar' object has no attribute 'instance'**`
1b. if an attribute `self.np_true = np.bool_(True)` is defined outside the kernel, the kernel expression  `self.np_true == True` raises an `AssertionError` in the compiler.

2. Comparing two numpy bools produces the same error as 1b. 
Example: `self.np_true==self.np_false`, where the attributes were defined outside the kernel  by:
```python
self.np_true = np.bool_(True)
self.np_false = np.bool_(False)
```

3. Casting a numpy to python bool in the kernel returns incorrect results. With `self.np_false` as defined in 2, calling `print(self.np_false)` from the kernel prints `False`. However, `print(bool(self.np_false))` prints `True`.
4. Using numpy bool in an if statement always evaluates as `True`.
Example: (using `self.np_false` as defined in 2)
```python
if self.np_false:
    print("Incorrect False evaluation")
# prints: "Incorrect False evaluation"
```

Similar to [Issue 736](https://github.com/m-labs/artiq/issues/736) bool casting to int in the kernel is not possible for numpy bools.

This behavior means that to do logic with arrays one needs to use  arrays with `dtype=np.object_`.
### Expected Behavior

I would expect numpy bool and python bool to be treated the same by the compiler. Numpy False should not evaluate to True. Minimally, this behavior should be documented.

whitequark, 2019-02-12:
I wasn't aware that `numpy.bool_` exists when implementing compiler support for numpy. (Actually, why does it exist?)

airwoodix, 2019-02-13:
`__builtins__.bool` is an integer subclass. `numpy.bool_` is a true (sic) boolean type. This changes the behavior of bitwise not. This is a nice summary I think: https://joergdietrich.github.io/python-numpy-bool-types.html

hartytp, 2019-02-13:
@airwoodix thanks for posting the nice summary.

@whitequark I believe the original reason this came up was trying to use numpy arrays of bools. As @pathfinder49 points out, currently one either has to to use `dtype=_object` in the array, or use a python list (which is not generally recommended practice), which isn't obvious from the docs.

commit 1f40f3ce (2021-03-20): compiler: Map host numpy.bool_ values to TBool

Since we don't implement any integer-like operations for TBool
(addition, bitwise not, etc.), TBool is currently neither
strictly equivalent to builtin bool nor numpy.bool_, but through
very obvious compiler errors (operation not supported) rather than
silently different runtime behaviour.

Just mapping both to TBool thus is a huge improvement over the
current behaviour (where numpy.False_ is a true-like object). In
the future, we could still implement more operations for TBool,
presumably following numpy.bool_ rather than the builtin type,
just  [... cut]
commit 9e3b6fac (2021-03-21): compiler: Map host numpy.bool_ values to TBool

Since we don't implement any integer-like operations for TBool
(addition, bitwise not, etc.), TBool is currently neither
strictly equivalent to builtin bool nor numpy.bool_, but through
very obvious compiler errors (operation not supported) rather than
silently different runtime behaviour.

Just mapping both to TBool thus is a huge improvement over the
current behaviour (where numpy.False_ is a true-like object). In
the future, we could still implement more operations for TBool,
presumably following numpy.bool_ rather than the builtin type,
just  [... cut]
closed 2021-03-20 by dnadlinger