m-labs/artiq #1276: ARTIQ Manual: getting started with core language, code comment error

RNourshargh, 2019-02-13:
# Bug Report

## One-Line Summary
Error in a code comment in the ARTIQ Manual: "getting started with core language"

The final code snippet incorrectly states the playback advance as 50*(100+100)ns. It should say 100*(100+100)ns 

## Issue Details
The final code block in "Direct Memory Access" includes the comments:
           "# execute RTIO operations in the DMA buffer
            # each playback advances the timeline by 50*(100+100) ns  "
The second line of this is incorrect and should read:
       "  # each playback advances the timeline by 100*(100+100) ns "

Each loop has a duration of (100+100)ns. 100ns pulse followed by 100ns delay. 100 such loops are carried out, therefore the full playback advances the timeline by:
100*(100+100) ns

Relevant code:
"for i in range(100):
                self.ttl0.pulse(100*ns)
                delay(100*ns)"

For most ARTIQ users this is trivial, but as a novice ARTIQ/Python user this caused some confusion.

The page in question is here:
https://github.com/m-labs/artiq/blob/master/doc/manual/getting_started_core.rst

### Steps to Reproduce

### Expected Behavior

### Actual (undesired) Behavior

### Your System
NA, documentation error.

* Operating System:
* Conda version:
* ARTIQ version: (package or git commit id, versions for bitstream, bootloader, runtime and host software). Run `conda list`
* Hardware involved:

dnadlinger, 2019-02-13:
Fixed, thanks. By the way, for similar issues in the future, you can just click the pencil (edit) button on the page you linked, and submit a pull request straight from the web UI.

closed 2019-02-13 by dnadlinger

RNourshargh, 2019-02-13:
Thanks. Wasn't sure if that was good practice. If I come across any more like this I will.

commit 944a3b34 (2019-02-13): doc: Fix typo in getting_started_rtio

GitHub: Fixes #1276.