m-labs/artiq #1281: UART in ARTIQ

drewrisinger, 2019-02-21:
# Question

## Category: Gateware/Coredevices

## Description

**Is there any sort of UART/USART driver in ARTIQ?** I've looked through documentation and coredevices, but can't see anything similar. Currently trying to hack SPI to work as a UART, but it's not ideal (getting the delays right so no timing delay between writes is an issue, and the SPI buffer is too small to output the data words that I want at once). Writing something like [coredevice/shiftreg.py](https://github.com/m-labs/artiq/blob/master/artiq/coredevice/shiftreg.py) could be an option, but I get consistent RTIO Underflow errors when trying to send data at ~ 5 MHz (on kc705).

**Related question**: What is going on in the coredevice that it can't queue a FIFO buffer with ~150 GPIO/TTL toggles 20 seconds in the future (tested on ARTIQ 2, raises RTIOUnderflow)? Something like:

```python
@kernel
def run_kernel(self):
    self.core.reset()
    self.core.break_realtime()
    delay(20 * s)
    for _ in range(70):
        self.gpio.on()
        delay(100 * ns)
        self.gpio.off()
        delay(100 * ns)
```

My best understanding is that ARTIQ should compute the GPIO timeline during the 20 second delay, and then output it when the 20 seconds have elapsed. Is this correct? Would this work better on ARTIQ4 where the for-loop could theoretically be per-computed and replayed via DMA?

dnadlinger, 2019-02-21:
Regarding your related question, the output event FIFOs only have a certain depth, and when they are full, CPU execution blocks until there is space available again. In your example, the CPU would block for ~20 s in the first `on`/`off` call not to fit in the buffer, and then proceed to push events once they are drained, at which point it can't generate them quick enough. This would indeed work better with DMA, where the steady-state event throughput is higher.

jordens, 2019-02-21:
The gateware could start filling another lane but that would reduce the number of allowed event reorderings. IMO it would be ok to use up to half the number of lanes for this purpose.

jordens, 2019-02-21:
If you really need an RTIO uart, you can write one. There is a generic uart in misoc that could be adapted.

dhslichter, 2019-02-21:
@jordens if this was being done in ARTIQ 2, then there are no lanes -- this would just be something about the FIFO depths on the particular gateware being used, which IIRC might be either 64 or 128 events.  If 64, this could explain the underflow, especially since the delays are in nanoseconds so there is floating point arithmetic being done at two points for each loop iteration.

dhslichter, 2019-02-22:
You can test this by making your loop do only 64 (or fewer) iterations instead of 70, and see if you still get underflows.

jordens, 2019-02-22:
Yes. My suggestion applies to sed. Buy it's worth considering there.

closed 2019-02-22 by jordens