m-labs/artiq #1282: Gateware: Increase Lanes

drewrisinger, 2019-02-21:
# Question

## Category: Gateware/RTIO

## Description

[Sequence Errors in manual](https://m-labs.hk/artiq/manual-master/rtio.html?highlight=lanes#sequence-errors) describes being able to add more lanes to the gateware. Looking through the gateware config [kc705.py](https://github.com/m-labs/artiq/blob/master/artiq/gateware/targets/kc705.py) doesn't seem to have any references to it. Where is such a config setting, if it exists?

dnadlinger, 2019-02-21:
See the `lane_count` parameter to `artiq.gateware.rtio.core.Core` (which is passed down the tree to `SED`, etc.). Could use some documentation, though.

dhslichter, 2019-02-22:
@drewrisinger is this for ARTIQ 2 as you mentioned in #1281 ?  There are no lanes in ARTIQ 2.

closed 2019-03-14 by sbourdeauducq