m-labs/artiq #1285: timing constraints not met in sayma masterdac build

vmsch, 2019-02-27:
I get a timing failed error in the build of the standard sayma masterdac 
artiq.gateware.targets.sayma_amc -V masterdac

`python -m artiq.gateware.targets.sayma_amc -V masterdac --rtm-csr-csv artiq_sayma/rtm_gateware/rtm_csr.csv --without-sawg`

This is with vivado 2018.3 (but the same error occurred also with 2018.1) and with artiq git 
commit ec230d6.

The timing summary in top_timing.rpt is

```
------------------------------------------------------------------------------------------------
| Design Timing Summary
| ---------------------
------------------------------------------------------------------------------------------------

    WNS(ns)      TNS(ns)  TNS Failing Endpoints  TNS Total Endpoints      WHS(ns)      THS(ns)  THS Failing Endpoints  THS Total Endpoints     WPWS(ns)     TPWS(ns)  TPWS Failing Endpoints  TPWS Total Endpoints  
    -------      -------  ---------------------  -------------------      -------      -------  ---------------------  -------------------     --------     --------  ----------------------  --------------------  
     -2.531       -5.051                      2                93143        0.030        0.000                      0                93083        0.122        0.000                       0                 35827  

Timing constraints are not met.

```

And the full file:
[top_timing.txt](https://github.com/m-labs/artiq/files/2910914/top_timing.txt)

sbourdeauducq, 2019-02-27:
I don't have time to debug this right now, please use the satellite variant.

vmsch, 2019-02-27:
Ok, the satellite builds fine, and we are happy to use only that for now.

sbourdeauducq, 2019-04-12:
@vmsch Do you have any use case for the masterdac variant or can we remove it?

vmsch, 2019-04-12:
We only used the masterdac for debugging, and can do everything with the satellite, so you can remove it.

sbourdeauducq, 2019-04-12:
OK, same here, it's only something I did for debugging - but it turns out using Kasli (+Urukul) as master instead is much simpler.

closed 2019-06-14 by sbourdeauducq