m-labs/artiq #1286: invalid HMC7043 identification

vmsch, 2019-02-28:
I receive an error 

```
[     8.289000s] ERROR(board_artiq::hmc830_7043::hmc7043): invalid HMC7043 ID: 0x00000000
panic at src/libcore/result.rs:945:5: cannot initialize HMC830/7043: "invalid HMC7043 identification"

```
when starting the Sayma. This was run with Sayma satellite gateware (commit ec230d6) flashed on, with --no-sawg option, but also occurs with sawg, and the same error occurred with two different RTM boards. The LED on the Kasli indicating established DRTIO link remains off.

Sometimes the lock establishes successfully but about a second afterwards I then receive another error:

```
[    17.315265s] ERROR(satman): failed to align SYSREF at FPGA: DDMTD helper PLL failed to lock 
[    22.447969s] ERROR(satman): received packet of an unknown type
```

This error occurred first, after flashing the Sayma with gateware built with vivado 2018.3. Previously, with gateware built with 2018.1 the Sayma started up correctly and put out Rf.
However, after the first occurrence of this error with the new gateware, this error now also occurs after flashing on the old gateware. The only difference I am aware of is that the Kasli master still has the new gateware on, because we don't have a copy of the old one. But I am not sure how that could cause such an error.

**Usual case:**

```
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2019 M-Labs Limited

Bootloader CRC passed
Gateware ident 5.0.dev+1089.g02a88d54;satellite.without-sawg
Initializing SDRAM...
DQS initial delay: 110 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010100000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010101001000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101110000000000000000000000000000000000000000000000000000000000
DQS initial delay: 110 taps
Write leveling: 105 116 143 134 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 227+-85 218+-93 199+-85 183+-93 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000006s]  INFO(satman): ARTIQ satellite manager starting...
[     0.004752s]  INFO(satman): software ident 5.0.dev+1089.g02a88d54;satellite.without-sawg
[     0.012842s]  INFO(satman): gateware ident 5.0.dev+1089.g02a88d54;satellite.without-sawg
[     0.020902s]  INFO(board_artiq::slave_fpga): Loading slave FPGA gateware...
[     0.027852s]  INFO(board_artiq::slave_fpga):   magic: 0x5352544d, length: 0x000c447c
[     1.056374s]  INFO(board_artiq::slave_fpga):   ...done
[     1.060118s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     1.094848s]  INFO(board_artiq::serwb):  ...done.
[     1.098158s]  INFO(board_artiq::serwb): RTM to AMC link test...
[     2.580461s]  INFO(board_artiq::serwb):   ...passed
[... 91 lines cut ...]
[    15.821050s]  INFO(board_artiq::ad9154): AD9154-0 initializing... 
[    15.828394s]  INFO(board_artiq::ad9154):   ...done 
[    15.912814s]  INFO(board_artiq::ad9154): AD9154-1 initializing... 
[    15.922313s]  INFO(board_artiq::ad9154):   ...done 
[    15.996113s]  INFO(board_artiq::ad9154): AD9154-1 running PRBS test... 
[    17.002393s]  INFO(board_artiq::ad9154):   ...passed 
[    17.005979s]  INFO(board_artiq::ad9154): AD9154-1 running STPL test... 
[    17.012765s]  INFO(board_artiq::ad9154):   c0 errors: 0 
[    17.017973s]  INFO(board_artiq::ad9154):   c1 errors: 0 
[    17.023183s]  INFO(board_artiq::ad9154):   c2 errors: 0 
[    17.028393s]  INFO(board_artiq::ad9154):   c3 errors: 0 
[    17.033316s]  INFO(board_artiq::ad9154):   ...passed 
[    17.048301s]  INFO(board_artiq::ad9154): AD9154-1 initializing... 
[    17.055646s]  INFO(board_artiq::ad9154):   ...done 
[    17.311750s]  INFO(satman): TSC loaded from uplink 
[    17.315265s] ERROR(satman): failed to align SYSREF at FPGA: DDMTD helper PLL failed to lock 
[    17.323521s]  INFO(board_artiq::jesd204sync): using DAC SYSREF delay from config: 13 
[    17.331213s]  INFO(board_artiq::jesd204sync): verifying SYSREF margins at DAC-0... 
[    17.359427s] ERROR(board_artiq::jesd204sync):   rotation at delay=15 is 2 delay steps from target (FAIL) 
[    17.367516s] ERROR(satman): failed to align SYSREF at DAC: insufficient SYSREF margin at DAC 
[    20.744707s]  INFO(satman): uplink is down, switching to local crystal clock 
[    20.779036s]  INFO(board_artiq::si5324): waiting for Si5324 lock... 
[    22.444397s]  INFO(board_artiq::si5324):   ...locked 
[    22.447969s] ERROR(satman): received packet of an unknown type 
```

sbourdeauducq, 2019-03-01:
> This error occurred first, after flashing the Sayma with gateware built with vivado 2018.3. Previously, with gateware built with 2018.1 the Sayma started up correctly and put out Rf.
> However, after [...]

Yes, Sayma is prone to this kind of behavior, which is one main reason why development has been so painfully slow and frustrating. Hopefully v2 boards will be better.

There is no "lock" since the HMC7043 is not a PLL. My guess is that the HMC7043 reset rework on your boards is a bit flaky.

> The LED on the Kasli indicating established DRTIO link remains off.

This sounds normal, if clocking initialization fails then DRTIO transceivers aren't clocked properly.

> `[ 17.315265s] ERROR(satman): failed to align SYSREF at FPGA: DDMTD helper PLL failed to lock `

Did you follow those instructions? https://github.com/m-labs/artiq/issues/1260

vmsch, 2019-03-01:
> My guess is that the HMC7043 reset rework on your boards is a bit flaky.

Thanks. We probed the reset pin on the HMC7043 and it starts high and turns low at the 
`[     2.583944s]  INFO(board_artiq::serwb): AMC to RTM link test...`
stage in the Sayma startup, however in the code the only place this pin is set to toggle is at the
`enabling HMC7043`
stage a few seconds later.

Would this and the fact that we see the same error on two independent boards maybe suggest it is more likely to be a software than a hardware issue?

> Did you follow those instructions? #1260

I haven't implemented these yet, but I can try that next. However I guess that shouldn't affect the first error, only the second one?

sbourdeauducq, 2019-03-01:
> Would this and the fact that we see the same error on two independent boards maybe suggest it is more likely to be a software than a hardware issue?

I also have boards where this never happens...

> I haven't implemented these yet, but I can try that next. However I guess that shouldn't affect the first error, only the second one?

Yes. And the second error will systematically occurs unless those instructions are followed.

closed 2019-08-15 by hartytp