m-labs/artiq #1304: Compiler: Array Declaration uses references

drewrisinger, 2019-04-05:
# Bug Report

## One-Line Summary

Issues with references when allocating multi-dimensional array on core.

## Issue Details

### Steps to Reproduce

Sample code:
```python
import artiq
from artiq.language import kernel

class TwoDimensionalArrayTest(artiq.language.environment.EnvExperiment):
    """Check if 2-D arrays can be created & manipulated on core."""

    def build(self):
        """Declare needed devices."""
        self.setattr_device("core")

    @kernel
    def run(self):
        """Run tests on 2D arrays."""
        self.test_python_array()    # DOES NOT WORK
        self.test_python_array_2()
        
    def test_python_array(self):
        """Test using an allocated multiplied array (DOES NOT WORK).

        This seems to have some sort of problem with referencing.
        """
        # ALLOCATION DOES NOT WORK
        x_len = 10
        y_len = 5
        arr = [[0] * x_len] * y_len

        counter = 0
        for i in range(y_len):
            for j in range(x_len):
                arr[i][j] = counter
                counter += 1

        print("Python method 1 (BROKEN)")
        print(arr)

    @kernel
    def test_python_array_2(self):
        """Test using an explicitly allocated array (WORKS)."""
        x_len = 10
        y_len = 5
        arr = [[0 for i in range(x_len)] for j in range(y_len)]

        counter = 0
        for i in range(y_len):
            for j in range(x_len):
                arr[i][j] = counter
                counter += 1

        print("Python method 2")
        print(arr)
```
### Expected Behavior

Both methods should do produce exactly the same behavior

### Actual (undesired) Behavior

* broken method produces same list of 10 items 5x (40-49), correct method counts 1-49.
* print: 
```
Python method 1 (BROKEN)
[[40, 41, 42, 43, 44, 45, 46, 47, 48, 49], [40, 41, 42, 43, 44, 45, 46, 47, 48, 49], [40, 41, 42, 43, 44, 45, 46, 47, 48, 49], [40, 41, 42, 43, 44, 45, 46, 47, 48, 49], [40, 41, 42, 43, 44, 45, 46, 47, 48, 49]]
Python method 2
[[0, 1, 2, 3, 4, 5, 6, 7, 8, 9], [10, 11, 12, 13, 14, 15, 16, 17, 18, 19], [20, 21, 22, 23, 24, 25, 26, 27, 28, 29], [30, 31, 32, 33, 34, 35, 36, 37, 38, 39], [40, 41, 42, 43, 44, 45, 46, 47, 48, 49]]
```

### Your System

* Operating System: Linux
* ARTIQ version: 4.0 (on kc705)
* Version of the gateware and runtime loaded in the core device: gateware: 4.0.dev+1838.gb56c7cec, software: 4.0+48.g01213e23.

whitequark, 2019-04-05:
This is the correct behavior. If you run your "broken" `test_python_array` method on the host Python, you will get the exact same result.

closed 2019-04-05 by whitequark

drewrisinger, 2019-04-14:
Fine. That seems like an inconsistent use of pass-by-reference/pass-by-value in the Python interpreter, though.

sbourdeauducq, 2019-04-14:
It is consistent (even if perhaps confusing) - you just have to know that the ``*`` operator on a list does a shallow copy of the element. Does the code below make it clearer?
```
>>> x = [0]
>>> id(x)
140512082107016
>>> l=[x]*2
>>> l
[[0], [0]]
>>> l[0][0] = 1
>>> l
[[1], [1]]
>>> id(l[0])
140512082107016
>>> id(l[1])
140512082107016
```