m-labs/artiq #1307: TTL and DDS panels not showing up on Linux

philipkent, 2019-04-16:
# Bug Report

## One-Line Summary
The TTL and DDS panels have disappeared for our version of Artiq (3.6) running on XUbuntu.  

## Issue Details

Both of the DDS and TTL panels were visible when we first installed the Artiq environment, but they have both recently disappeared. 

### Steps to Reproduce
It is random/intermittent, so this is not known. 

### Your System

* Operating System: XUbuntu version 18.04, Bionic Beaver (Xfce version 4.12)
* ARTIQ version: 4.6
* Version of the gateware and runtime loaded in the core device: 4.6
* If using Conda, output of `conda list`:

aiohttp                   3.4.4            py35h470a237_0    conda-forge/label/main
artiq                     3.6             py_0+git394b66cd    m-labs/label/main
artiq-kc705-nist_qc2      3.6             py_0+git394b66cd    m-labs/label/main
async-timeout             3.0.1                   py_1000    conda-forge/label/main
asyncserial               0.1             py_13+git340e430    m-labs/label/main
attrs                     19.1.0                     py_0    conda-forge/label/main
binutils-or1k-linux       2.30                          7    m-labs/label/main
blas                      2.4                    openblas    conda-forge/label/main
bzip2                     1.0.6             h14c3975_1002    conda-forge/label/main
ca-certificates           2019.3.9             hecc5488_0    conda-forge/label/main
certifi                   2018.8.24             py35_1001    conda-forge/label/main
cffi                      1.11.5           py35h5e8e0c9_1    conda-forge/label/main
chardet                   3.0.4                    py35_3    conda-forge/label/main
dbus                      1.13.0            h4e0c4b3_1000    conda-forge/label/main
expat                     2.2.5             hf484d3e_1002    conda-forge/label/main
fontconfig                2.13.1            he4413a7_1000    conda-forge/label/main
freetype                  2.10.0               he983fc9_0    conda-forge/label/main
gettext                   0.19.8.1          hc5be6a0_1002    conda-forge/label/main
glib                      2.56.2            had28632_1001    conda-forge/label/main
gst-plugins-base          1.12.5            h3865690_1000    conda-forge/label/main
gstreamer                 1.12.5            h0cc0488_1000    conda-forge/label/main
h5py                      2.8.0            py35h7eb728f_2    conda-forge/label/main
hdf5                      1.10.2               hc401514_3    conda-forge/label/main
icu                       58.2              hf484d3e_1000    conda-forge/label/main
idna                      2.7                      py35_2    conda-forge/label/main
idna_ssl                  1.0.0                         0    conda-forge/label/main
jpeg                      9c                h14c3975_1001    conda-forge/label/main
levenshtein               0.12.0                   py35_1    m-labs/label/main
libblas                   3.8.0                4_openblas    conda-forge/label/main
libcblas                  3.8.0                4_openblas    conda-forge/label/main
libffi                    3.2.1             he1b5a44_1006    conda-forge/label/main
libgcc-ng                 8.2.0                hdf63c60_1  
libgfortran               3.0.0                         1    conda-forge/label/main
libgfortran-ng            7.3.0                hdf63c60_0  
libgit2                   0.24.1                        7    m-labs/label/main
libiconv                  1.15              h516909a_1005    conda-forge/label/main
liblapack                 3.8.0                4_openblas    conda-forge/label/main
liblapacke                3.8.0                4_openblas    conda-forge/label/main
libopenblas               0.3.3                h5a2b251_3  
libpng                    1.6.36            h84994c4_1000    conda-forge/label/main
libssh2                   1.7.0.git                     5    m-labs/label/main
libstdcxx-ng              8.2.0                hdf63c60_1  
libusb                    1.0.20                        0    m-labs/label/main
libuuid                   2.32.1            h14c3975_1000    conda-forge/label/main
libxcb                    1.13              h14c3975_1002    conda-forge/label/main
libxml2                   2.9.8             h143f9aa_1005    conda-forge/label/main
lit                       0.4.1                      py_9    m-labs/label/main
llvmlite-artiq            0.20.0                   py35_1    m-labs/label/main
multidict                 4.4.2            py35h470a237_0    conda-forge/label/main
ncurses                   6.1               hf484d3e_1002    conda-forge/label/main
numpy                     1.15.2           py35h99e49ec_0  
numpy-base                1.15.2           py35h2f8d375_0  
openblas                  0.3.5             h9ac9557_1001    conda-forge/label/main
openocd                   0.10.0                        1    m-labs/label/main
openssl                   1.0.2r               h14c3975_0    conda-forge/label/main
outputcheck               0.4.2                      py_7    m-labs/label/main
pcre                      8.41              hf484d3e_1003    conda-forge/label/main
pip                       18.0                  py35_1001    conda-forge/label/main
prettytable               0.7.2                      py_2    conda-forge/label/main
pthread-stubs             0.4               h14c3975_1001    conda-forge/label/main
pycparser                 2.19                       py_0    conda-forge/label/main
pygit2                    0.24.0                   py35_4    m-labs/label/main
pyqt                      5.6.0            py35h8210e8a_7    conda-forge/label/main
pyqtgraph                 0.10.0                     py_5    conda-forge/label/main
pyserial                  3.4                        py_2    conda-forge/label/main
python                    3.5.5                h5001a0f_2    conda-forge/label/main
python-dateutil           2.8.0                      py_0    conda-forge/label/main
pythonparser              1.1                        py_8    m-labs/label/main
qt                        5.6.2                hf70d934_9    conda-forge/label/main
quamash                   0.5.5                      py_4    m-labs/label/main
readline                  7.0               hf8c457e_1001    conda-forge/label/main
regex                     2015.11.22               py35_1    m-labs/label/main
scipy                     1.1.0            py35he2b7bc3_1  
setuptools                40.4.3                   py35_0    conda-forge/label/main
sip                       4.18.1           py35hfc679d8_0    conda-forge/label/main
six                       1.11.0                   py35_1    conda-forge/label/main
sqlite                    3.26.0            h67949de_1001    conda-forge/label/main
system                    5.8                           2  
tk                        8.6.9             h84994c4_1000    conda-forge/label/main
wheel                     0.32.0                py35_1000    conda-forge/label/main
xorg-libxau               1.0.9                h14c3975_0    conda-forge/label/main
xorg-libxdmcp             1.1.3                h516909a_0    conda-forge/label/main
xz                        5.2.4             h14c3975_1001    conda-forge/label/main
yarl                      1.2.6            py35h470a237_0    conda-forge/label/main
zlib                      1.2.11            h14c3975_1004    conda-forge/label/main

* Hardware involved: N/A

dnadlinger, 2019-04-16:
You shouldn't even be able to disable them (save for a bug, of course) – if you right-click the dock on the right-hand side of the main window to get the list of enabled docks, are TTL/DDS listed there?

philipkent, 2019-04-16:
@dnadlinger they are listed, but the check marks are missing next to their entries.

philipkent, 2019-04-16:
I just tried running the dashboard under a fresh conda environment with artiq v3.6 installed and it has the same problem.  I'm wondering it it has something to due with the config file for the dashboard since it looks like environments share the same file.  Maybe it has become corrupted.

sbourdeauducq, 2019-04-17:
Probably. Make a backup of it, post it in this issue, and then try deleting it.

philipkent, 2019-04-25:
We were able to fix this by first closing the dashboard, then deleting the `artiq_dashboard.pyon` file, and finally restarting the dashboard.  That created a new config file and restored both of the DDS and TTL panels.  On our version of XUbuntu the location of the file was `~/.config/artiq/3/artiq_dashboard.pyon`

@sbourdeauducq Here is the contents of our old `artiq_dashboard.pyon` file that was causing the TTL and DDS panels to not appear.  Thanks.

{
    "LogDockManager": {"log0": {
        "header": b'\x00\x00\x00\xff\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05!\x00\x00\x00\x02\x01\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00d\xff\xff\xff\xff\x00\x00\x00\x81\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\xb1\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x04p\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x03\xe8\x00',
        "min_level_idx": 1,
        "freetext_filter": ""
    }},
    "AppletsCCBDock": {
        "ccbp_global": "create",
        "applets": [("applet", 0, true, "Current Scan", {
            "command": "$python -m artiq_ions.applets.plot_xy\ncurrent_scan.plots.y\n--x current_scan.plots.x\n--fit current_scan.plots.fitline\n--title current_scan.plots.plot_title\n--x_scale current_scan.plots.x_scale\n--x_units current_scan.plots.x_units\n--y_scale current_scan.plots.y_scale\n--x_label current_scan.plots.x_label\n--y_units current_scan.plots.y_units\n--y_label current_scan.plots.y_label\n--trigger 1   --rid current_scan.rid\n",
            "ty": "command"
        }, null), ("applet", 1, false, "Dimension 0 Scan", {
            "command": "$python -m artiq_ions.applets.plot_xy\ncurrent_scan.plots.y\n--x current_scan.plots.x\n--fit current_scan.plots.fitline\n--title current_scan.plots.plot_title\n--x_scale current_scan.plots.x_scale\n--x_units current_scan.plots.x_units\n--x_label current_scan.plots.x_label\n--y_units current_scan.plots.y_units\n--y_label current_scan.plots.y_label\n--trigger current_scan.plots.trigger\n",
            "ty": "command"
        }, b'\x01\xd9\xd0\xcb\x00\x02\x00\x00\x00\x00\x02\x1a\x00\x00\x05;\x00\x00\x04/\x00\x00\x06\xc3\x00\x00\x02\x1a\x00\x00\x05;\x00\x00\x04/\x00\x00\x06\xc3\x00\x00\x00\x01\x00\x00\x00\x00\x07\x80'), ("applet", 2, true, "Counts", {
            "command": "${artiq_applet}big_number counts",
            "ty": "command"
        }, null), ("applet", 3, true, "Dimension 1 Scan", {
            "command": "$python -m artiq_ions.applets.plot_xy\ncurrent_scan.plots.dim1.y\n--x current_scan.plots.dim1.x\n--fit current_scan.plots.dim1.fitline\n--title current_scan.plots.dim1.plot_title\n--x_scale current_scan.plots.dim1.x_scale\n--x_units current_scan.plots.dim1.x_units\n--x_label current_scan.plots.dim1.x_label\n--y_units current_scan.plots.dim1.y_units\n--y_label current_scan.plots.dim1.y_label\n--trigger current_scan.plots.dim1.trigger",
            "ty": "command"
        }, null), ("applet", 4, true, "Current Histogram", {
            "command": "$python -m artiq_ions.applets.plot_hist\ncurrent_hist.bins\n--x current_hist.bin_boundaries\n--x_units current_hist.x_units\n--x_label current_hist.x_label\n--plot_title current_hist.plot_title",
            "ty": "command"
        }, null), ("applet", 5, false, "Heating Rate", {
            "command": "$python -m applets.plot_xy_ntrace_white\n\nheating_rate.bsb.stats.mean\n--x1 heating_rate.bsb.stats.points\n--error1 heating_rate.bsb.stats.error\n\nheating_rate.rsb.stats.mean\n--x2 heating_rate.rsb.stats.points\n--error2 heating_rate.rsb.stats.error\nheating_rate.bkgd.stats.mean\n--x3 heating_rate.bkgd.stats.points\n--error3 heating_rate.bkgd.stats.error\n--title \"RSB, BSB, Bkgd vs Wait Time\"",
            "ty": "command"
        }, b'\x01\xd9\xd0\xcb\x00\x02\x00\x00\x00\x00\x04\xf0\x00\x00\x04z\x00\x00\x07K\x00\x00\x06\x05\x00\x00\x04\xf0\x00\x00\x04z\x00\x00\x07K\x00\x00\x06\x05\x00\x00\x00\x01\x00\x00\x00\x00\x07\x80')]
    },
    "ExplorerDock": {"current_directory": "/home/stylab/artiq/repository/drivers/bkp/"},
    "ExperimentManager": {
        "arguments": {
            "repo:startup/Alignment": OrderedDict([("shelving_on", {
                "state": false,
                "tooltip": null,
                "group": "Detection",
                "desc": {
                    "default": false,
                    "ty": "BooleanValue"
                }
            }), ("repeats", {
                "state": 500,
                "tooltip": null,
                "group": null,
                "desc": {
                    "default": 100,
                    "unit": "",
                    "ty": "NumberValue",
                    "max": null,
                    "ndecimals": 0,
                    "min": null,
                    "step": 10,
                    "scale": 1.0
                }
            })]),
            "repo:scans/Shim Scan": OrderedDict([("shim", {
                "state": "y",
                "tooltip": null,
                "group": null,
                "desc": {
                    "default": "x",
                    "choices": ["x", "y", "z", "xcom", "ycom", "dc2", "dc3", "dc4"],
                    "ty": "EnumerationValue"
                }
            }), ("npasses", {
                "state": 1,
                "tooltip": null,
[... 4631 lines cut ...]
                "hdf5_load_directory": "/home/stylab"
            }
        }
    },
[... cut ...]
        "geometry": b'\x01\xd9\xd0\xcb\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\x7f\x00\x00\x047\x00\x00\x00\x00\x00\x00\x00\x18\x00\x00\x07\x7f\x00\x00\x047\x00\x00\x00\x00\x02\x00\x00\x00\x07\x80'
    },
    "ShortcutsDock": {
        0: "",
        1: "repo:startup/Ion Monitor",
        2: "",
        3: "repo:startup/Alignment",
        4: "repo:calibrations/Calibrate Microwaves",
        5: "repo:calibrations/Calibrate Raman",
        6: "repo:scans/Heating Rate Scan",
        7: "",
        8: "repo:ResetSidebandFreqs",
        9: "",
        10: "",
        11: "repo:startup/Load Ion"
    }
}

sbourdeauducq, 2019-06-13:
@philipkent Please do not paste such a huge file that makes the issue unreadable. Use the file attachment feature instead.

Anyway, the lines that make the panel disappear are those:
```
"MainWindow": {
        "geometry": b'\x01\xd9\xd0\xcb\x00\x03\x00\x00\x00\x00\x03|\x00\x00\x00\x00\x00\x00\x08\x0f\x00\x00\x04t\x00\x00\x03\x80\x00\x00\x00 \x00\x00\x08\x0b\x00\x00\x04p\x00\x00\x00\x00\x00\x00\x00\x00\n\x00\x00\x00\x03\x80\x00\x00\x00 \x00\x00\x08\x0b\x00\x00\x04p',
        "state": b'\x00\x00\x00\xff\x00\x00\x00\x00\xfd\x00\x00\x00\x02\x00\x00\x00\x01\x00\x00\x02\x00\x00\x00\x03D\xfc\x02\x00\x00\x00\x07\xfc\x00\x00\x00\x00\x00\x00\x03D\x00\x00\x01\xd1\x01\x00\x00\x1b\xfa\x00\x00\x00\x00\x02\x00\x00\x00\x08\xfb\x00\x00\x00\x0e\x00A\x00p\x00p\x00l\x00e\x00t\x00s\x01\x00\x00\x00\x00\xff\xff\xff\xff\x00\x00\x00v\x01\x00\x00\x03\xfb\x00\x00\x00\x10\x00S\x00c\x00h\x00e\x00d\x00u\x00l\x00e\x01\x00\x00\x00\x00\xff\xff\xff\xff\x00\x00\x00v\x01\x00\x00\x03\xfb\x00\x00\x00\x10\x00E\x00x\x00p\x00l\x00o\x00r\x00e\x00r\x01\x00\x00\x00\x00\xff\xff\xff\xff\x00\x00\x00\xc0\x01\x00\x00\x03\xfb\x00\x00\x00\x10\x00D\x00a\x00t\x00a\x00s\x00e\x00t\x00s\x01\x00\x00\x00\x00\xff\xff\xff\xff\x00\x00\x00\xa7\x01\x00\x00\x03\xfb\x00\x00\x00\x12\x00S\x00h\x00o\x00r\x00t\x00c\x00u\x00t\x00s\x01\x00\x00\x00\x00\xff\xff\xff\xff\x00\x00\x01\xb5\x01\x00\x00\x03\xfb\x00\x00\x00\x06\x00T\x00T\x00L\x02\x00\x00\x00\t\x00\x00\x04\xe2\x00\x00\x02\x9d\x00\x00\x00\xbe\xfb\x00\x00\x00\x06\x00D\x00D\x00S\x02\x00\x00\x00\x11\x00\x00\x05&\x00\x00\x02\xad\x00\x00\x00z\xfb\x00\x00\x00\x06\x00D\x00A\x00C\x01\x00\x00\x00\x00\xff\xff\xff\xff\x00\x00\x00"\x01\x00\x00\x03\xfb\x00\x00\x00\x0e\x00a\x00p\x00p\x00l\x00e\x00t\x004\x00\x00\x00\x03O\x00\x00\x01\x02\x00\x00\x00\x00\x00\x00\x00\x00\xfb\x00\x00\x00\x0e\x00a\x00p\x00p\x00l\x00e\x00t\x005\x02\x00\x00\x04\xf0\x00\x00\x04z\x00\x00\x02\\\x00\x00\x01\x8c\xfb\x00\x00\x00\x0e\x00a\x00p\x00p\x00l\x00e\x00t\x001\x02\x00\x00\x02\x1a\x00\x00\x05;\x00\x00\x02\x16\x00\x00\x01\x89\xfb\x00\x00\x00\x0e\x00a\x00p\x00p\x00l\x00e\x00t\x000\x02\x00\x00\x06\x96\x00\x00\x00\x00\x00\x00\x00\xf0\x00\x00\x00\x96\xfb\x00\x00\x00\x0e\x00a\x00p\x00p\x00l\x00e\x00t\x003\x02\x00\x00\x07\x86\x00\x00\x00\x00\x00\x00\x00\xf0\x00\x00\x00\x96\xfb\x00\x00\x00\x0e\x00a\x00p\x00p\x00l\x00e\x00t\x002\x02\x00\x00\x02\x8c\x00\x00\x03"\x00\x00\x02\x00\x00\x00\x00k\x00\x00\x00\x03\x00\x00\x04\x8c\x00\x00\x01\x0c\xfc\x01\x00\x00\x00\x01\xfb\x00\x00\x00\x08\x00l\x00o\x00g\x000\x01\x00\x00\x00\x00\x00\x00\x04\x8c\x00\x00\x01j\x01\x00\x00\x03\x00\x00\x02\x8b\x00\x00\x03D\x00\x00\x00\x04\x00\x00\x00\x04\x00\x00\x00\x08\x00\x00\x00\x08\xfc\x00\x00\x00\x00'
    },
```

In particular, it is the ``state`` entry (which is passed directly into ``QtWidgets.QMainWindow.restoreState``) that hides those panels.

A workaround is to remove the ``MainWindow`` entry from ``artiq_dashboard.pyon`` when the problem occurs.

sbourdeauducq, 2019-06-13:
For a better fix, I would need to know how the state got corrupted in the first place and why it seems to affect only those two panels. Could be a Qt bug and/or something weird with what we're doing in those panels.

commit 8853cf8d (2019-06-13): dashboard: work around disappearing TTL/DDS panel bug. Closes #1307
commit 5fe37cad (2019-06-14): dashboard: work around disappearing TTL/DDS panel bug. Closes #1307
closed 2019-06-13 by sbourdeauducq