m-labs/artiq #1313: TTL Input Overflows

connorgoham, 2019-05-07:
# Bug Report

## One-Line Summary

Calling timestamp_mu() for several TTL input channels results in RTIO input overflows, even for a small number of input events.

## Issue Details
Not sure if this is a bug or if I am misunderstanding something about how input events are recorded, but calling timestamp_mu() for multiple input channels leads to overflow errors for a sparse number of input events. How many input events are the TTL channels meant to handle before an overflow? 

### Steps to Reproduce

code:
```
from artiq.experiment import *

class TTLInFIFOTest(EnvExperiment):
	"""Test TTL In FIFO capacity"""

	def build(self):
		self.setattr_device("core")
		self.setattr_device("scheduler")
		self.setattr_device("ttl0")
		self.setattr_device("ttl1")

		self.detect_t = 1.0
		

	@kernel
	def run(self):
		self.core.reset()
		self.ttl0.gate_rising(self.detect_t)
		detection_t1 = self.ttl0.timestamp_mu(now_mu())
		detection_t2 = self.ttl1.timestamp_mu(now_mu())
		count0 = self.ttl0.count(now_mu())
```

TTL Inputs:

- greater than 66 Hz square wave (50% duty cycle, 0 V min, 5 V max) input on ttl0 causes overflows, for  less than 65 Hz the code runs without reporting an error

- nothing input on ttl1

### Expected Behavior

I expect the code to run without an RTIO input overflow, at least for this frequency of input events.

### Actual (undesired) Behavior

For a square wave on TTL0 with a frequency greater than 66 Hz there is an error:
`root:Terminating with exception (RTIOOverflow: RTIO input overflow on channel 0)`

Commenting out either `detection_t2 = self.ttl1.timestamp_mu(now_mu())` or `count0 = self.ttl0.count(now_mu())` and the code runs without an error.

### Your System

* Operating System: Ubuntu 18.04.2 LTS
* ARTIQ version: 4.0 release
* Version of the gateware and runtime loaded in the core device: 
-`software ident 4.0;opticlock`
-`gateware ident 4.0;opticlock`
* If using Conda, output of `conda list`:
```
# packages in environment at /home/rabi/anaconda3:
#
# Name                    Version                   Build  Channel
_ipyw_jlab_nb_ext_conf    0.1.0            py36he11e457_0  
alabaster                 0.7.10           py36h306e16b_0  
anaconda                  5.2.0                    py36_3  
anaconda-client           1.6.14                   py36_0  
anaconda-navigator        1.8.7                    py36_0  
[... 230 lines cut ...]
tornado                   5.0.2                    py36_0  
traitlets                 4.3.2            py36h674d592_0  
typing                    3.6.4                    py36_0  
unicodecsv                0.14.1           py36ha668878_0  
unixodbc                  2.3.6                h1bed415_0  
urllib3                   1.22             py36hbe7ace6_0  
wcwidth                   0.1.7            py36hdf4376a_0  
webencodings              0.5.1            py36h800622e_1  
werkzeug                  0.14.1                   py36_0  
wheel                     0.31.1                   py36_0  
widgetsnbextension        3.2.1                    py36_0  
wrapt                     1.10.11          py36h28b7045_0  
xlrd                      1.1.0            py36h1db9f0c_1  
xlsxwriter                1.0.4                    py36_0  
xlwt                      1.3.0            py36h7b00a1f_0  
xz                        5.2.4                h14c3975_4  
yaml                      0.1.7                had09818_2  
zeromq                    4.2.5                h439df22_0  
zict                      0.1.3            py36h3a3bf81_0  
zlib                      1.2.11               ha838bed_2  
```

* Hardware involved:
-Kasli v1.1
-EEM Octal rev1.1

jordens, 2019-05-07:
Your code never get to counting the other events on `ttl0` since it blocks on waiting for an event on `ttl1` which never happens, hence the overflow once the buffer is full after 1s*65Hz - 1 = 64 events.

jordens, 2019-05-07:
And opticlock is typically built for Kasli v1.0. That may harm your hardware if you didn't change that.

connorgoham, 2019-05-07:
Ah OK I was misunderstanding what was blocking and when. To be clear, for a piece of code like:
`self.ttl0.gate_rising(detect_t)`
`detect_end = now_mu()`
`self.ttl0.count(detect_end)`
the `count()` method doesn't first block until detect_end and then count all events in the buffer, but rather it actively tries to count events in the buffer and only blocks as long as there are no further events to empty from the buffer and the wall clock < detect_end. Is this a correct way of looking at it?