m-labs/artiq #1322: artiq=5.0.dev+378.gbcda53ee PackagesNotFoundError

Whichislove, 2019-05-15:
# Question
We can't install the needed version of Artiq.

## Category: installation/setup

## Description

Question text
We have received Sinara hardware and just getting started using it. In the message from Robert Jordens we got "But you don't have to install target binaries at all currently. Your
system has everything already flashed onto it. To get started you will
just need artiq (conda install artiq=5.0.dev=378.gbcda53ee if you want
to match the exact
version)."

So we tried to install it on Windows 64.
I guess that there should be "conda install artiq=5.0.dev+378.gbcda53ee" instead of equal sign. Am I right?

Anyway I tried to use both with "=" and "+" signs but it outputs the following error:
PackagesNotFoundError: The following packages are not available from current channels:

Please find attached screen of the Error. I am also including conda info as it might be usefull.
 [image]

Then I tried installing on Linux.
I was able to install artiq pakages via conda on linux. I checked that I can ping IP that was indicated in device_db.py file that we received from you.

However, when I tired to execute "artiq_run led.py"
the output was an error saying that Connection refused. Similar error about connection refuse occured when we followed installation procedures on WIndows according to manual.

My suspicion is that we need to flash binaries to our system as we are installing the latest version following manual. We avoided flashing step as it was mentioned in Robert's message that everything is already flashed to avoid messing up things.

[image]

As we are just setting up we also would like to know which OS Windows or Linux is preferred according to the feedback you got from the customers?

Thanks in advance, sorry for many questions. We need to make the system working ASAP and unfortunately none of us is an expert yet.

sbourdeauducq, 2019-05-15:
> However, when I tired to execute "artiq_run led.py"
> the output was an error saying that Connection refused. Similar error about connection refuse occured when we followed installation procedures on WIndows according to manual.

Did you set the correct IP address in the device database?

Whichislove, 2019-05-15:
We got device_db.py file with core_addr = "10.0.16.129". It was preset by Robert. Wireless connection IPv4 was changed manually to the same 10.0.16.129

Following manual I tired to ping it and it worked. It sent and received several packages without any loss.

sbourdeauducq, 2019-05-15:
> Wireless connection IPv4 was changed manually to the same 10.0.16.129

This is wrong.

sbourdeauducq, 2019-05-15:
> Following manual I tired to ping it and it worked. It sent and received several packages without any loss.

You are just pinging your computer, not the device.

Whichislove, 2019-05-15:
The reason why I set IPv4 manually is that when it was automatic, connection error was popping up.

Could you please tell me what should be done to connect to the Kasli system? 
We are using TP-LINK MC220L that was suggested by Robert to connect via optical fiber.

sbourdeauducq, 2019-05-15:
Setting the IP address manually is fine but setting the same IP address as the device is not.

Whichislove, 2019-05-15:
I tried to set it manually to 192.168.1.75 as in manual but still getting the same Error. Connection refused.

sbourdeauducq, 2019-05-15:
Please familiarize yourself with IP networking concepts instead of just trying random values.
Setting 10.0.16.1 has a higher chance of working.

Whichislove, 2019-05-15:
10.0.16.1 resulted in "Network is unreachable"

sbourdeauducq, 2019-05-15:
https://www.cisco.com/en/US/docs/security/vpn5000/manager/reference/guide/appA.html

sbourdeauducq, 2019-05-15:
And, is the Ethernet link up on the Kasli and the switch/network card it is connected to (link LEDs on)?

Whichislove, 2019-05-15:
Yes, LEDs on Kasli and switch are on

Whichislove, 2019-05-16:
Thanks for a useful link. I was able to connect to Kasli and played with LED turning on and off. However, there is a WARNING about mismatch between gateware (5.0.dev+378.gbcda53ee) and software (5.6857.d9fc2847.beta) versions. 

Do we need to do anything with this?

sbourdeauducq, 2019-05-17:
You're should be fine for now - no need to reflash - but if you want to fix that warning later, locate the firmware variant that corresponds to your system (I don't know which one it is) on https://git.m-labs.hk/m-labs/sinara-systems and follow the reflashing instructions in the manual.

closed 2019-05-17 by sbourdeauducq