m-labs/artiq #1325: compiler: Slices with negative stride are broken

dnadlinger, 2019-05-19:
Slices with negative stride don't seem to work at all on master ().

See the below test case – the first assert works, but the other slices seem to end up empty.
```py
# RUN: %python -m artiq.compiler.testbench.jit %s
# RUN: %python %s

data = [0, 1, 2, 3]

assert data[:] == data

# FIXME: Type inference for the integer literal fails inside the slice.
m1 = -1
assert data[::m1] == [3, 2, 1, 0]
assert data[:0:m1] == [3, 2, 1]

m2 = -2
assert data[m2::m1] == [2, 1, 0]
```

(The type inference issue seems unrelated.)

@whitequark: Does this immediately ring a bell?

whitequark, 2019-05-20:
Yes. There are no tests for negative strides, and I suspect I completely overlooked that case while writing that code. (I think I may have been unaware that a negative stride is legal at the time, but I'm not sure, it was a while ago.)

sbourdeauducq, 2022-01-13:
Fixed in NAC3:

```python
@extern
def output_int32(x: int32):
    ...

@extern
def output_asciiart(x: int32):
    ...

def output_list(l: list[int32]):
    for x in l:
        output_int32(x)
    output_asciiart(-1)

def run() -> int32:
    data = [0, 1, 2, 3]
    output_list(data[:])

    m1 = -1
    output_list(data[::m1])
    output_list(data[:0:m1])

    m2 = -2
    output_list(data[m2::m1])
    return 0
```

```
0
1
2
3

3
2
1
0

3
2
1

2
1
0

```

closed 2022-01-13 by sbourdeauducq