m-labs/artiq #1326: changes to the release process

sbourdeauducq, 2019-05-21:
Starting with ARTIQ-5 I would like to see changes to the release process (which are already partially implemented):
* Call "development" versions "beta" versions. This is mostly to avoid confusion with development environments; for example, some people thought that the ``artiq-dev`` package was the development version of ARTIQ.
* Move to a rolling release process. Enough of this tedious and error-prone tagging/rebuilding that causes delays for users to see bugfixes, and causes frustrating issues for us such a FPGA bitstreams passing timing with the candidate version but not after the version string has changed after tagging.
* The latest stable version is the latest commit that passed tests on the stable branch. Ditto with the beta branch. The output from each branch goes in separate channels on the build server.
* The beta branch is ``master``, and the current stable branch is a branch defined in ``nix-scripts``.
* This makes cutting a new release very easy (create a branch, update the ``master`` major version, and update ``nix-scripts`` to point to the new stable branch), as opposed to the current process.
* We can also introduce prerelease builds: an optional third channel that would be set to the "stable" branch so that we can keep working on it and get CI, before updating the stable channel for users.
* The version numbers are formatted as ``<major version>.<distance to root commit>.<short git hash>[.beta]``

dhslichter, 2019-05-21:
Sounds reasonable to me.  I haven't followed the `nix-scripts` shift but it appears that this handles conda builds as well as nix builds, correct?

sbourdeauducq, 2019-05-21:
It also generates the conda packages, yes.

sbourdeauducq, 2019-08-08:
Add another ``oldstable`` channel so that the current ``stable`` builds do not disappear immediately when a new release is made.

drewrisinger, 2019-11-05:
> It also generates the conda packages, yes.

FWIW, the M-Labs Conda channel hasn't been updated with new files in 6 months: https://anaconda.org/m-labs/artiq/files. Or is there a new place where these conda packages live now?

sbourdeauducq, 2019-11-06:
https://conda.m-labs.hk/artiq-beta/

sbourdeauducq, 2019-11-06:
Though if you are looking for build information then check https://nixbld.m-labs.hk/

closed 2019-11-14 by sbourdeauducq

sbourdeauducq, 2020-08-31:
Also add a tag where stable rolling releases start.