m-labs/artiq #1331: artiq_dashboard won't start. Possible issue with PyQt5 installation?

RNourshargh, 2019-06-11:
# Bug Report

## One-Line Summary
artiq_dashboard won't start. 

Short summary.
The artiq_dashboard command fails to start the gui and throws an error message. artiq_run on individual experiments still functions correctly as does artiq_client submit.

## Issue Details
During the last artiq install there were error messages related to the packages which it cannot find.

### Steps to Reproduce
$ artiq_dashboard

### Expected Behavior
Dashboard gui opens

Behavior

### Actual (undesired) Behavior

$source activate artiq

(artiq)
$ artiq_dashboard
Traceback (most recent call last):
  File "C:\Users\RJN651\AppData\Local\Continuum\anaconda3\envs\artiq\Scripts\artiq_dashboard-script.py", line 5, in <module>
    from artiq.frontend.artiq_dashboard import main
  File "C:\Users\RJN651\AppData\Local\Continuum\anaconda3\envs\artiq\lib\site-packages\artiq\frontend\artiq_dashboard.py", line 9, in <module>
    from PyQt5 import QtCore, QtGui, QtWidgets
ImportError: DLL load failed: The specified module could not be found.

* Text description
* Log message, tracebacks, screen shots where relevant

### Your System

* Operating System: windows 10 Enterprise N version 1709 
* ARTIQ version: (with recent versions of ARTIQ, run ``artiq_client --version``)
ARTIQ v5.6852.c2b4f0cf.beta

* Version of the gateware and runtime loaded in the core device: (in the output of ``artiq_coremgmt -D .... log``)
I couldn't figure out what syntax was needed to make this work, but as the issue is confined to the PC I don't think this is neccessary. If I need to find this I will need some more advice on exactly what I need to enter into the command line.

* If using Nix, version of nixpkgs (run ``nix-instantiate --eval -E '(import <nixpkgs> {}).lib.version'``):
* If using Conda, output of `conda list`:
$ conda list
WARNING conda.core.prefix_data:_load_single_record(167): Ignoring malformed prefix record at: C:\Users\RJN651\AppData\Local\Continuum\anaconda3\envs\artiq\conda-meta\artiq-board-kasli-mitll.json
WARNING conda.core.prefix_data:_load_single_record(167): Ignoring malformed prefix record at: C:\Users\RJN651\AppData\Local\Continuum\anaconda3\envs\artiq\conda-meta\artiq-board-kasli-tester.json
WARNING conda.core.prefix_data:_load_single_record(167): Ignoring malformed prefix record at: C:\Users\RJN651\AppData\Local\Continuum\anaconda3\envs\artiq\conda-meta\artiq-board-kc705-nist_clock.json
WARNING conda.core.prefix_data:_load_single_record(167): Ignoring malformed prefix record at: C:\Users\RJN651\AppData\Local\Continuum\anaconda3\envs\artiq\conda-meta\artiq.json
# packages in environment at C:\Users\RJN651\AppData\Local\Continuum\anaconda3\envs\artiq:
#
# Name                    Version                   Build  Channel
aiohttp                   3.4.4            py35hfa6e2cd_0    conda-forge
artiq                     5.6852.c2b4f0cf.beta          pypi_0    pypi
async-timeout             3.0.1                   py_1000    conda-forge
attrs                     19.1.0                     py_0    conda-forge
binutils-or1k-linux       2.27                 h93a10e1_6    m-labs
blas                      1.0                         mkl
bscan-spi-bitstreams      0.10.0                        2    m-labs
ca-certificates           2019.3.9             hecc5488_0    conda-forge
certifi                   2018.8.24             py35_1001    conda-forge
cffi                      1.11.5           py35hfa6e2cd_1    conda-forge
chardet                   3.0.4                    py35_3    conda-forge
curl                      7.64.0               h4496350_2    conda-forge
h5py                      2.8.0            py35hb24199d_2    conda-forge
hdf5                      1.10.2               he51fdeb_2    conda-forge
icc_rt                    2019.0.0             h0cc432a_1
icu                       58.2                     vc14_0    conda-forge
idna                      2.7                      py35_2    conda-forge
idna_ssl                  1.0.0                         0    conda-forge
intel-openmp              2019.3                      203
jpeg                      8d                       vc14_2
krb5                      1.16.3            hdd46e55_1001    conda-forge
levenshtein               0.12.0                   py35_1    m-labs
libcurl                   7.64.0               h4496350_2    conda-forge
libgit2                   0.27.8               hfac1375_0    conda-forge
libiconv                  1.15              hfa6e2cd_1005    conda-forge
libpng                    1.6.37               h7602738_0    conda-forge
libssh2                   1.8.2                h642c060_2    conda-forge
libtiff                   4.0.6                    vc14_2
libusb                    1.0.20                        0    m-labs
llvmlite-artiq            0.23.0.dev               py35_5    m-labs
mkl                       2018.0.3                      1
mkl_fft                   1.0.9                    py35_0    conda-forge
mkl_random                1.0.1                    py35_0    conda-forge
multidict                 4.4.2            py35hfa6e2cd_0    conda-forge
numpy                     1.15.2           py35ha559c80_0
numpy-base                1.15.2           py35h8128ebf_0
openocd                   0.10.0                        6    m-labs
openssl                   1.1.1b               hfa6e2cd_2    conda-forge
pip                       18.0                  py35_1001    conda-forge
prettytable               0.7.2                      py_2    conda-forge
pycparser                 2.19                       py_0    conda-forge
pygit2                    0.27.1           py35hfa6e2cd_0    conda-forge
pyqt                      4.11.4                   py35_2    conda-forge
pyqt5                     5.5.1                    py35_0    m-labs
pyqtgraph                 0.10.0                     py_5    conda-forge
pyreadline                2.1                   py35_1000    conda-forge
python                    3.5.5                he025d50_2    conda-forge
python-dateutil           2.8.0                      py_0    conda-forge
pythonparser              1.1                        py_8    m-labs
qt                        4.8.7                    vc14_6
qt5                       5.5.1                         1    m-labs
quamash                   0.5.5                      py_3    m-labs
regex                     2018.08.29       py35hfa6e2cd_0    conda-forge
scipy                     1.1.0            py35h4f6bf74_1
setuptools                40.4.3                   py35_0    conda-forge
sip                       4.18                     py35_1    conda-forge
six                       1.11.0                   py35_1    conda-forge
sqlite                    3.26.0            hfa6e2cd_1001    conda-forge
tk                        8.6.9             hfa6e2cd_1001    conda-forge
vc                        14                            0    conda-forge
vs2015_runtime            14.0.25420                    0    conda-forge
wheel                     0.32.0                py35_1000    conda-forge
wincertstore              0.2                   py35_1002    conda-forge
yarl                      1.2.6            py35hfa6e2cd_0    conda-forge
zlib                      1.2.11            h2fa13f4_1004    conda-forge

* Hardware involved:
The PC running artiq. This error is present without any connected core device.

sbourdeauducq, 2019-06-11:
How did you install ARTIQ?

RNourshargh, 2019-06-11:
Most recently I used the ARTIQ installer script ($ python install-with-conda.py).  I think it returned some errors on the install but I didn't investigate these at the time because artiq_run was working.
Many Thanks

sbourdeauducq, 2019-06-11:
What error was it?
Can you try with the same anaconda we use for testing? https://repo.anaconda.com/archive/Anaconda3-2019.03-Windows-x86_64.exe (https://git.m-labs.hk/M-Labs/nix-scripts/src/branch/master/artiq/windows/install.nix#L14)

RNourshargh, 2019-06-12:
This was the last part of the terminal output when I did the install: (sorry for the screen dump)

The following NEW packages will be INSTALLED:

  aiohttp            conda-forge/win-64::aiohttp-3.4.4-py35hfa6e2cd_0
  artiq              tmpm9uoj37r/noarch::artiq-5.6852.c2b4f0cf.beta-0
  async-timeout      conda-forge/noarch::async-timeout-3.0.1-py_1000
  attrs              conda-forge/noarch::attrs-19.1.0-py_0
  binutils-or1k-lin~ m-labs/win-64::binutils-or1k-linux-2.27-h93a10e1_6
  blas               pkgs/main/win-64::blas-1.0-mkl
  bscan-spi-bitstre~ m-labs/noarch::bscan-spi-bitstreams-0.10.0-2
  ca-certificates    conda-forge/win-64::ca-certificates-2019.3.9-hecc5488_0
  cffi               conda-forge/win-64::cffi-1.11.5-py35hfa6e2cd_1
  chardet            conda-forge/win-64::chardet-3.0.4-py35_3
  curl               conda-forge/win-64::curl-7.64.0-h4496350_2
  h5py               conda-forge/win-64::h5py-2.8.0-py35hb24199d_2
  hdf5               conda-forge/win-64::hdf5-1.10.2-he51fdeb_2
  icc_rt             pkgs/main/win-64::icc_rt-2019.0.0-h0cc432a_1
  icu                conda-forge/win-64::icu-58.2-vc14_0
  idna               conda-forge/win-64::idna-2.7-py35_2
  idna_ssl           conda-forge/noarch::idna_ssl-1.0.0-0
  intel-openmp       pkgs/main/win-64::intel-openmp-2019.3-203
  jpeg               conda-forge/win-64::jpeg-9c-hfa6e2cd_1001
  krb5               conda-forge/win-64::krb5-1.16.3-h038dc86_1000
  levenshtein        m-labs/win-64::levenshtein-0.12.0-py35_1
  libcurl            conda-forge/win-64::libcurl-7.64.0-h4496350_2
  libgit2            conda-forge/win-64::libgit2-0.27.8-hfac1375_0
  libiconv           conda-forge/win-64::libiconv-1.15-hfa6e2cd_1005
  libpng             conda-forge/win-64::libpng-1.6.37-h7602738_0
  libssh2            conda-forge/win-64::libssh2-1.8.0-hc4dcbb0_1003
  libusb             m-labs/win-64::libusb-1.0.20-0
  llvmlite-artiq     m-labs/win-64::llvmlite-artiq-0.23.0.dev-py35_5
  mkl                pkgs/main/win-64::mkl-2018.0.3-1
  mkl_fft            conda-forge/win-64::mkl_fft-1.0.9-py35_0
  mkl_random         conda-forge/win-64::mkl_random-1.0.1-py35_0
  multidict          conda-forge/win-64::multidict-4.4.2-py35hfa6e2cd_0
  numpy              pkgs/main/win-64::numpy-1.15.2-py35ha559c80_0
  numpy-base         pkgs/main/win-64::numpy-base-1.15.2-py35h8128ebf_0
  openocd            m-labs/win-64::openocd-0.10.0-6
  openssl            conda-forge/win-64::openssl-1.0.2r-hfa6e2cd_0
  prettytable        conda-forge/noarch::prettytable-0.7.2-py_2
  pycparser          conda-forge/noarch::pycparser-2.19-py_0
  pygit2             conda-forge/win-64::pygit2-0.27.1-py35hfa6e2cd_0
  pyqt               conda-forge/win-64::pyqt-5.6.0-py35h764d66f_7
  pyqtgraph          conda-forge/noarch::pyqtgraph-0.10.0-py_5
  pyreadline         conda-forge/win-64::pyreadline-2.1-py35_1000
  python-dateutil    conda-forge/noarch::python-dateutil-2.8.0-py_0
  pythonparser       m-labs/noarch::pythonparser-1.1-py_8
  qt                 conda-forge/win-64::qt-5.6.2-h2639256_8
  quamash            m-labs/noarch::quamash-0.5.5-py_4
  regex              conda-forge/win-64::regex-2018.08.29-py35hfa6e2cd_0
  scipy              pkgs/main/win-64::scipy-1.1.0-py35h4f6bf74_1
  sip                conda-forge/win-64::sip-4.18.1-py35h6538335_0
  six                conda-forge/win-64::six-1.11.0-py35_1
  sqlite             conda-forge/win-64::sqlite-3.26.0-hfa6e2cd_1001
  tk                 conda-forge/win-64::tk-8.6.9-hfa6e2cd_1001
  yarl               conda-forge/win-64::yarl-1.2.6-py35hfa6e2cd_0
  zlib               conda-forge/win-64::zlib-1.2.11-h2fa13f4_1004

Downloading and Extracting Packages
ca-certificates-2019 | 184 KB    | ############################################################################ | 100%
scipy-1.1.0          | 13.5 MB   | ############################################################################ | 100%
icu-58.2             | 21.8 MB   | #############################5                                               |  39%
pyqtgraph-0.10.0     | 480 KB    | ############################################################################ | 100%
libpng-1.6.37        | 1.3 MB    | ############################################################################ | 100%
pygit2-0.27.1        | 646 KB    | ############################################################################ | 100%
openssl-1.0.2r       | 5.4 MB    | ############################################################################ | 100%
tk-8.6.9             | 3.7 MB    | ############################################################################ | 100%
yarl-1.2.6           | 122 KB    | ############################################################################ | 100%
artiq-5.6852.c2b4f0c | 560 KB    | ############################################################################ | 100%
[... 23 lines cut ...]
numpy-1.15.2         | 48 KB     | ############################################################################ | 100%
attrs-19.1.0         | 32 KB     | ############################################################################ | 100%
libgit2-0.27.8       | 650 KB    | ############################################################################ | 100%
bscan-spi-bitstreams | 1.4 MB    | ############################################################################ | 100%
sqlite-3.26.0        | 958 KB    | ############################################################################ | 100%
[... cut ...]
libssh2-1.8.0        | 184 KB    | ############################################################################ | 100%

('Connection broken: OSError("(10054, \'WSAECONNRESET\')")', OSError("(10054, 'WSAECONNRESET')"))
('Connection broken: OSError("(10054, \'WSAECONNRESET\')")', OSError("(10054, 'WSAECONNRESET')"))
('Connection broken: OSError("(10054, \'WSAECONNRESET\')")', OSError("(10054, 'WSAECONNRESET')"))

command 'conda install -y -n artiq-main -c C:\Users\RJN651\AppData\Local\Temp\tmpm9uoj37r artiq' returned non-zero exit
status: 1

RNourshargh, 2019-06-12:
In order to use this alternative version of anaconda do I ned to uninstall my existing version? Will I need to recreate my environments?

sbourdeauducq, 2019-06-12:
> In order to use this alternative version of anaconda do I ned to uninstall my existing version? Will I need to recreate my environments?

I don't know.

> The following NEW packages will be INSTALLED:

One thing that looks suspicious is it mentions simultaneous installation of different versions of Qt5 and PyQt5:
 qt5 5.5.1 1 m-labs
 qt conda-forge/win-64::qt-5.6.2-h2639256_8
 pyqt5 5.5.1 py35_0 m-labs
 pyqt conda-forge/win-64::pyqt-5.6.0-py35h764d66f_7

Try cleaning that up maybe (e.g. uninstall one of the two Qt/Pyqt pairs).

RNourshargh, 2019-06-13:
Thanks for the advice. I used conda to remove all of the above and then reinstalled pyqt5 and qt5 with conda. 
$ conda install pyqt5
This didn't make any difference.

Next I tried using pip install for pyqt5 , quamash, and pyqtgraph

I did this one package at a time and used the error message to choose which package was next. The artiq dashboard now loads and appears to be running correctly. I am not sure if using pip install will cause further problems down the line but for now it appears to have worked.

Should I close the issue?

sbourdeauducq, 2019-06-13:
If pip works, then that should be alright by conda standards.
If you notice issues, you can alternatively try installing the other Qt packages (qt=5 and pyqt=5) with conda.

closed 2019-06-13 by sbourdeauducq

RNourshargh, 2019-06-13:
Ok brilliant. Thank you for all of your help