m-labs/artiq #1334: Pushing 2 git repositories to the bare

RNourshargh, 2019-06-21:
# Question

Is there a way to push experiments from 2 git repositories to the artiq dashboard?  

## Category: FILL_IN

## Description
I have made 2 git repositories to store my artiq code, one public and one private. The private repository (stored on the one computer currently running artiq) is integrated with artiq (pushing to a bare repository as per the instructions) and committed, pushed experiments appear in the dashboard and run as expected. I am hoping that I can push experiments from my second repository (this one public and containing simple examples to share with fellow beginners) to the dashboard as well but don't know how to do this. I tried importing the public repo as a submodule in the private repo but the experiments didn't appear even after the commit was pushed. Any advice on this would be greatly apreciated.

sbourdeauducq, 2019-06-21:
No, this isn't implemented. But you can tell the master to use one repository or the other using its command-line arguments.

RNourshargh, 2019-06-21:
Ok. Thank you. I've just managed to get that working.

closed 2019-06-21 by RNourshargh