m-labs/artiq #1336: artiq conda packages contain malformed metadata

jordens, 2019-07-04:
```
WARNING conda.core.prefix_data:_load_single_record(180): Ignoring malformed prefix record at: C:\ProgramData\Miniconda3\envs\artiq-5-3\conda-meta\artiq-board-kasli-ptb3.json
WARNING conda.core.prefix_data:_load_single_record(180): Ignoring malformed prefix record at: C:\ProgramData\Miniconda3\envs\artiq-5-3\conda-meta\artiq.json
```

sbourdeauducq, 2019-07-04:
Does not happen on CI e.g. https://nixbld.m-labs.hk/build/8066/log
```
<3>Microsoft Windows [Version 10.0.17763.107]
<3>Copy /nix/store/9jzgb6zslzdf4sxsvbi0jv60cww22zi8-conda-artiq/noarch/artiq-5.6898.4d01410c.beta-0.tar.bz2 to artiq.tar.bz2
<3>ssh windows anaconda\scripts\activate artiq-env && conda install artiq.tar.bz2
<3>
<3>Downloading and Extracting Packages
<3>########## | 100%
<3>Preparing transaction: ...working... done
<3>Verifying transaction: ...working... done
<3>Executing transaction: ...working... done
<3>ssh windows shutdown -s -t 600
<3>ssh windows anaconda\scripts\activate artiq-env && set ARTIQ_ROOT=%cd%\anaconda\envs\artiq-env\Lib\site-packages\artiq\examples\kc705_nist_clock&&python -m unittest discover -v artiq.test.coredevice
<3>test_init (artiq.test.coredevice.test_ad9910.AD9910Test) ... skipped "test device not available: `Failed to get description of device 'urukul_ad9910'`"
...
```
Maybe that happens with some combination of conda-build and conda versions?

jordens, 2019-07-04:
This is a fresh and clean miniconda installation from today.

sbourdeauducq, 2019-07-05:
I've updated Anaconda/conda-build (note how easy this is with the new system), does the issue still exist?

sbourdeauducq, 2019-07-17:
Did not see this issue while testing https://github.com/m-labs/artiq/issues/1335#issuecomment-512367060

jordens, 2019-07-17:
Ok. I didn't get the chance to check again. Will close and reopen if it occurs.

closed 2019-07-17 by jordens

steine01, 2020-11-09:
I installed ARTIQ-6 (version  6.7480.a97b4633.beta) today with the `artiq-board-kasli-ptb3` firmware (build 100211 from https://nixbld.m-labs.hk/job/artiq/full-beta/conda-artiq-board-kasli-ptb3). When I execute` conda list` I get the same error as in this issue:
```
(artiq-6-1) C:\Users\ybion443>conda list
WARNING conda.core.prefix_data:_load_single_record(190): Ignoring malformed prefix record at: C:\Users\ybion443\.conda\envs\artiq-6-1\conda-meta\artiq-board-kasli-ptb3.json
# packages in environment at C:\Users\ybion443\.conda\envs\artiq-6-1:
#
# Name                    Version                   Build  Channel
aiohttp                   3.7.2            py37h2bbff1b_1
artiq                     6.7480.a97b4633.beta               0    https://conda.m-labs.hk/artiq-beta
artiq-board-kasli-ptb3    6.7480.a97b4633.beta               0    https://conda.m-labs.hk/artiq-beta
...
```
I don't know if it is connected, but I also always get the warning, that the old gateware is still installed, when executing the idle experiment:
```
artiq.dashboard.experiments:Submitted 'repo:Idle', RID is 22763
artiq.coredevice.comm_kernel:Mismatch between gateware (5.7112.d6aeb038) and software (6.7480.a97b4633.beta) versions
```
The flash seems to execute successfully:
```
(artiq-6-1) C:\Users\ybion443>artiq_flash -t kasli -V ptb3 -f flash_storage.img storage start
Open On-Chip Debugger 0.10.0-00013-gbb7bedad (2018-02-17-04:39)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 25000 kHz
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 25000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : gdb server disabled
TEMP 76.99 C
VCCINT 0.985 V
VCCAUX 1.775 V
VCCBRAM 0.991 V
VPVN 0.000 V
VREFP 0.000 V
VREFN 0.000 V
VCCPINT 0.000 V
VCCPAUX 0.000 V
VCCODDR 0.000 V
loaded file C:\Users\ybion443\.conda\envs\artiq-6-1\share\bscan-spi-bitstreams\bscan_spi_xc7a100t.bit to pld device 0 in 0s 237368us
Info : Found flash device 'sp s25fl128' (ID 0x00182001)
flash 'jtagspi' found at 0x00000000
Info : Found flash device 'sp s25fl128' (ID 0x00182001)
Info : sector 68 took 106 ms
erased sectors 68 through 68 on flash bank 1 in 0.106713s
Info : Found flash device 'sp s25fl128' (ID 0x00182001)
wrote 49 bytes from file flash_storage.img to flash bank 1 at offset 0x00440000 in 0.008975s (5.332 KiB/s)
Info : Found flash device 'sp s25fl128' (ID 0x00182001)
read 49 bytes from file flash_storage.img and flash bank 1 at offset 0x00440000 in 0.005986s (7.994 KiB/s)
contents match
```

sbourdeauducq, 2020-11-09:
> artiq_flash -t kasli -V ptb3 -f flash_storage.img storage start

This command is not supposed to flash the firmware. Please read the manual for the correct command.

sbourdeauducq, 2020-11-09:
As for the warning, it again does not occur on CI:
https://nixbld.m-labs.hk/build/100428/log

We have Anaconda 2020.02 on our system:
https://git.m-labs.hk/M-Labs/wfvm/src/branch/master/wfvm/layers/default.nix#L11

What version do you have?

sbourdeauducq, 2020-11-09:
Though CI doesn't install board packages on windows and it seems this is where the warning is from...

steine01, 2020-11-09:
> 
> 
> > artiq_flash -t kasli -V ptb3 -f flash_storage.img storage start
> 
> This command is not supposed to flash the firmware. Please read the manual for the correct command.

I am a bit confused, because in https://m-labs.hk/artiq/manual-beta/installing.html#installing-openocd there are the following commands as an example:
```
$ artiq_mkfs flash_storage.img -s mac xx:xx:xx:xx:xx:xx -s ip xx.xx.xx.xx
$ artiq_flash -t [board] -V [variant] -f flash_storage.img storage start
```

Additionally these already worked on another ARTIQ system we have. Are the arguments wrong?

> 
> 
> As for the warning, it again does not occur on CI:
> https://nixbld.m-labs.hk/build/100428/log
> 
> We have Anaconda 2020.02 on our system:
> https://git.m-labs.hk/M-Labs/wfvm/src/branch/master/wfvm/layers/default.nix#L11
> 
> What version do you have?

My output of `conda info` is:

```
(artiq-6-1) C:\Users\ybion443\src\generic-scpi-driver>conda info

     active environment : artiq-6-1
    active env location : C:\Users\ybion443\.conda\envs\artiq-6-1
            shell level : 2
       user config file : C:\Users\ybion443\.condarc
 populated config files : C:\Users\ybion443\.condarc
          conda version : 4.9.1
    conda-build version : 3.20.5
         python version : 3.6.12.final.0
       virtual packages : __win=0=0
                          __archspec=1=x86_64
       base environment : C:\ProgramData\Anaconda3  (read only)
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
                          https://conda.anaconda.org/quartiq/win-64
                          https://conda.anaconda.org/quartiq/noarch
                          https://conda.m-labs.hk/artiq-beta/win-64
                          https://conda.m-labs.hk/artiq-beta/noarch
                          https://conda.anaconda.org/m-labs/win-64
                          https://conda.anaconda.org/m-labs/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\ProgramData\Anaconda3\pkgs
                          C:\Users\ybion443\.conda\pkgs
                          C:\Users\ybion443\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\ybion443\.conda\envs
                          C:\ProgramData\Anaconda3\envs
                          C:\Users\ybion443\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.9.1 requests/2.24.0 CPython/3.6.12 Windows/10 Windows/10.0.17134
          administrator : False
             netrc file : None
           offline mode : False
```

sbourdeauducq, 2020-11-09:
See https://m-labs.hk/artiq/manual-beta/installing.html#writing-the-flash

> My output of `conda info` is:

I don't know how this corresponds to versions of the anaconda distribution. Anyway you probably can ignore the warning.

steine01, 2020-11-09:
> 
> 
> See https://m-labs.hk/artiq/manual-beta/installing.html#writing-the-flash
Ah I see thank you, the command without the flash storage worked correctly.