m-labs/artiq #1338: suservo: Core device driver should support/expose different timings

dnadlinger, 2019-07-07:
Currently, the servo cycle time is hard-coded in the `artiq.coredevice.suservo` driver.

`artiq.gateware.eem.SUServo.add_std` supports changing the servo loop timings via the `t_rtt` parameter. This is important when using Sampler in a remote crate, where the clock out -> clock/data in round trip time exceeds the default of 32 ns. If `t_rtt` is increased, though, the cycle time also changes, which throws off the I gain scaling and IIR update delay calculations.

The easiest way to fix this might be to just print the final computed cycle time during gateware builds and add a constructor argument to `SUServo` to be set accordingly from the device db.

For synchronising other operations to the servo cycle (cf. #1337), it might also be interesting to expose the different phases.

dnadlinger, 2019-07-08:
Actually, it seems like the comment in https://github.com/m-labs/artiq/blob/f7e10759dca56c09eae2723dd5baebc02542ce40/artiq/gateware/eem.py#L489-L490 is just wrong. Writing the AD9910 profile already takes 146 coarse clock cycles, so `t_rtt` needs to increase to 52 to impact the calculated cycle time. This is roughly a round-trip across 41 m at two thirds the speed of light, so unlikely to ever be (usefully) exceeded.

`T_CYCLE` in the core device driver is one cycle longer than the computed `t_cycle` in `gateware.servo.SUServo`, though, as it adds an extra cycle on top of the `t_dds` calculation. @jordens: Do you remember where that +1 came from? I'll add a comment either way…

dnadlinger, 2019-07-08:
(I'll commit this if my understanding is correct:
```diff
--- a/artiq/gateware/eem.py
+++ b/artiq/gateware/eem.py
@@ -487,7 +487,7 @@ class SUServo(_EEM):
             cabling delay and the 8 ns max propagation delay on Sampler (ADC
             and LVDS drivers). Increasing ``t_rtt`` increases servo latency.
             With all other parameters at their default values, ``t_rtt`` values
-            above 4 also increase the servo period (reduce servo bandwidth).
+            above 51 also increase the servo period (reduce servo bandwidth).
         :param clk: DDS SPI clock cycle half-width in RTIO coarse cycles
             (default: 1)
         :param shift: fixed-point scaling factor for IIR coefficients
```
.)

dnadlinger, 2019-07-08:
It also seems like the computation of the cycle length https://github.com/m-labs/artiq/blob/f7e10759dca56c09eae2723dd5baebc02542ce40/artiq/gateware/suservo/servo.py#L24
should actually be
```py
t_cycle = max(t_adc + 1, t_iir, t_dds)  # +1 for restart
```
as otherwise `t_restart > 1` isn't true.

jordens, 2019-07-16:
I counted the cycles in the simulation. On top of the `bits*period` time there should be one half cycle idle with CS asserted, and (at least) another half cycle with CS not asserted, plus the clock cycle it takes to transition between states (setting the `active` bits). Check with simulation, maybe I miscounted.

The `t_adc + 1` you propose is not correct AFAICT from a quick glance. If the ADC cycle time were to limit the overall cycle time, then the counter logic would needs to be changed.

dnadlinger, 2019-07-16:
I guess the easiest way of verifying this without a doubt would be to scan for the timestamps where `get_status()` bit 1 is being strobed at runtime. I was just comparing the eponymous values in gateware and driver; I might have indeed missed an extra cycle introduced by the FSM transition.

Do you agree that t_rtt < 52 shouldn't increase the total cycle time, though? (t_cycle doesn't change, then.)

Regarding t_adc, it might well be that there is more that needs changing; I didn't check. I just noticed that the code purports to handle arbitrary times, but doesn't actually do so.

jordens, 2019-07-16:
I am unsure about the max RTT. Make certain that the pipeline slots and the memory access by the different stages writer don't collide if you increase this. This (the 4) could be a vestigial number from when the pipeline looked different and both the ADC and the IIR stage needed to be shorter than the DDS stage.

c.f. also https://github.com/m-labs/artiq/issues/788#issuecomment-393578023

dnadlinger, 2019-07-16:
Turns out the cycle length is actually 146 * 8 = 1168 ns, not 147 periods, i.e. it matches the gateware t_cycle.

Determined using this quick hack (which could be added as a unit test, if we had suservo tests):

```py
from artiq.experiment import *
from functools import reduce
import math
import numpy as np

class DetermineServoPeriod(EnvExperiment):
    """Determine servo period

    This determines the length of a cycle on the given SUServo device by
    repeatedly fetchingthe status word and recording timestamps where the
    "done" bit is set (which is strobed for one clock cycle at the end of
    each cycle).
    """

    def build(self):
        self.servo_name = self.get_argument("Servo device name",
                                            StringValue(default="raom_suservo0"))

        self.num_dones = self.get_argument(
            "Number of 'done' observations",
            NumberValue(default=1000, min=2, step=1, ndecimals=0))

        self.setattr_device("core")

    def run(self):
        self.acquire(self.get_device(self.servo_name), self.num_dones)

    @kernel
    def acquire(self, servo, num_dones):
        timestamps = [np.int64(0)] * num_dones

        self.core.break_realtime()
        servo.set_config(enable=1)
        delay(10 * us)

        i = 0
        while i < num_dones:
            self.core.break_realtime()
            if servo.get_status() & 0x2:
                timestamps[i] = now_mu()
                i += 1

        self.post_results(timestamps)

    def post_results(self, timestamps):
        timestamps = np.array(timestamps)
        deltas = (timestamps[1:] - timestamps[:-1])

        cycle_mu = reduce(lambda a, b: math.gcd(a, b), deltas, deltas[0])
        print("Cycle length (or multiple thereof):",
            self.core.mu_to_seconds(cycle_mu), "s")
```

dnadlinger, 2019-07-16:
Regarding:

> Make certain that the pipeline slots and the memory access by the different stages writer don't collide if you increase this. 

Could you point me to an example of two possibly colliding accesses? I'm not sure I follow.

jordens, 2019-07-17:
There is an entire bench of unittests for suservo: https://github.com/m-labs/artiq/tree/master/artiq/gateware/test/suservo
HITL tests are #1004

jordens, 2019-07-17:
I was probably overly worried. One problem would have been the `x` value shifting phase of the IIR processor which runs in parallel with the DDS writes. But there is an assert protecting against that. In an older design there was also the possibility of the ADC `x` writes to memory colliding with the IIR `x` shifting and reads from memory. But since the shifting is now after the processing it's fine.

dnadlinger, 2019-07-17:
> There is an entire bench of unittests for suservo

Sure – I wasn't trying to be snarky, but merely wanted to mention that I was unsure about adding driver-side tests to the tree that then never actually get executed. I guess we could always feature-gate them, to make manually running them on hardware easier until automated tests are set up.

jordens, 2019-07-17:
Sure. There is tooling for gate these tests on suitability HITL device_db content already. I'm all for adding tests like these.