m-labs/artiq #1339: suservo: Support more than two Urukuls per Sampler

dnadlinger, 2019-07-07:
Currently, the SUServo gateware/driver assume that there are two Urukuls per servo instance, and one Sampler. This isn't unreasonable, as it corresponds to 8 ADC channels and 8 DDSes.

However, in typical experiments, many laser beamlines will end up having more than one wavelength/frequency component, and hence on average more than one DDS channel per ADC channel.

We already support choosing the source ADC channel for each profile, so support for an unequal number of ADC and DDS channels should not be too hard to add.

---

@jordens: Any comments regarding this? I had a quick look at splitting `w.channel` into `w.adc_channel` and `w.dds_channel`, and it doesn't seem too bad, but I'm not sure whether I've noticed all the intricacies of the pipelined design.

jordens, 2019-07-16:
This should be doable. It will lengthen the cycle accordingly but I don't see any road blocks.

dnadlinger, 2019-10-29:
We've got this implemented and working; need to find some time to clean up the changes.

dtcallcock, 2019-11-12:
Great to hear.  We definitely plan on using this.