m-labs/artiq #1340: HDF5 files are empty on Linux

TheCakeIsAPi, 2019-07-09:
# Bug Report

## One-Line Summary

HDF5 files are empty.

## Issue Details

This is seen on two linux installs in Monroe IonPhoton and Britton labs (ARTIQ 4.0 installed via cloning from github).  HDF5 files are okay on Monroe Euriqa install on Windows (ARTIQ 4.0 installed via conda).

I am able to set_dataset and mutate_dataset as expected.  The datasets show up in the Datasets tab in the Dashboard.  I can create applets and plot the datasets.  When broadcast=True the plots live update as expected.  The hdf5 files are created in the expected place in the results/date/hour directory.  They are not zero size, they are always 10.8 kB, they appear to be valid hdf5 files and can be opened without error in HDFView 3.1.0.  However, they have no internal tree structure, no internal datasets, no properties.

It should be the default, but I have also tried archive=True with no success.  save=True is deprecated but I have also tried that.  persist=true works as expected.  Also I have used artiq_run with the --hdf5 output.h5 flag, and had the same results (output.h5 is created but empty).

The results are the same whether or not set_dataset is in a @kernel or not, as well as whether or not only set_dataset is used or if mutate_dataset is also used.  I have tried both 1D and 2D datasets with the same results.

### Steps to Reproduce

Sample code:

from artiq.experiment import *
import time
import numpy as np

class dataset_test1(EnvExperiment):
    def build(self):

        self.setattr_device("core")

    @kernel
    def run(self):
        count=10
        self.set_dataset("parabola", np.full(count, np.nan), broadcast=True)
        for i in range(count):
            self.mutate_dataset("parabola", i, i*i)
            time.sleep(0.1)

### Your System

* Operating System:  Ubuntu 18.04.2 LTS 64-bit
* ARTIQ version: 4.0 (installed via clone from github)
* artiq_coremgmt log: 
INFO(runtime): software ident 4.0+0.gbf50dcf7.dirty;monroe_ionphoton
INFO(runtime): gateware ident 4.0+0.gbf50dcf7.dirty;monroe_ionphoton
* conda list: (see attached)
[conda_list.txt](https://github.com/m-labs/artiq/files/3374284/conda_list.txt)
* Hardware involved:  Kasli v1.0, Intel Core i7-8700K CPU

dnadlinger, 2019-07-09:
Just to be sure, what is the output of `h5dump` on those files? Also completely empty?

We (Oxford) are exclusively using Linux for all our main experiment masters, and haven't seen this across various Ubuntu releases.

dnadlinger, 2019-07-09:
I just checked, and one of our experiments happens to use the same`hdf5`/`h5py` versions as well.

TheCakeIsAPi, 2019-07-09:
Good thinking @dnadlinger .  h5dump shows that the files are not empty!

I have also found that if I make the dataset sufficiently large, the file size increases beyond 10.8 kB.  I had tried this before with a smaller change in dataset size, but it seems like there is some sort of chunking that makes the file take up this much space until you reach some threshold.

I am attaching one of my hdf5 files for reference, and I am going to try opening it on Windows (and vice versa).  It's not clear yet if this is an HDFView problem or not.

[000000684-dataset_test1.zip](https://github.com/m-labs/artiq/files/3374856/000000684-dataset_test1.zip)

TheCakeIsAPi, 2019-07-09:
Okay, so actually all the files (produced on both Linux and Windows) open just fine on Windows on HDFView 3.0.0, and all of them (produced on both Linux and Windows) appear empty when opened on Linux.  On Linux, I had thought I was running HDFView 3.1.0 (the latest drop), but I just noticed that the version installed via the Ubuntu Software app/store (and is also the same as installed via `sudo apt install hdfview`) is only 2.1.1.

So I guess we can close this and call it a problem with the older version of HDFView, although I find that a bit unsatisfying because 2.1.1 was mature enough, and these dataset files aren't very complicated.  Is there some newer hdf5 feature that the ARTIQ saved hdf5 files are using?  If so, perhaps we should consider not using that feature?

Also, @dnadlinger what HDF viewer do you use on Linux?  Mostly I use h5py to work with data, but it's nice to have a quick viewer.  I've now found that HDF Compass v.0.6.0 opens the files correctly.  HDFView 3.1.0 isn't available as a binary for Linux, did you compile it from the source?

TheCakeIsAPi, 2019-07-09:
Actually there's no need to install from source, the HDFView-3.1.0-centos7_64.tar.gz binary from https://www.hdfgroup.org/downloads/hdfview/ works fine on Ubuntu.  I didn't realize that before.  I can now open the saved hdf5 files and see all the datasets using HDFView 3.1.0 on Ubuntu.  I'm going to close the issue.  Thanks!

closed 2019-07-09 by TheCakeIsAPi