m-labs/artiq #1341: Voltage to MU isn't recognised as an attribute of zotino

cnourshargh, 2019-07-17:
`voltage_to_mu()`  isn't recognised as an attribute of the zotino. See my code below for outputting a ramp wave on the Zotino.

```
from artiq.experiment import *

class ZotinoOutputs(EnvExperiment):
    def build(self): 

        self.setattr_device("core")
        self.setattr_device("zotino0")

    @kernel
    def set_zotino_voltages(self, zotino, voltages):
        self.core.break_realtime()                          
        zotino.init()                                       
        delay(200*us)                                       

        while(1):                                           
            for voltage in voltages:                        
                    
                zotino.write_dac_mu(0, voltage)             
                zotino.load()                              
                delay(20*us)                               
                
    def test_zotinos(self):
        n_steps = 100                                      
        voltages = [0.1*i for i in range(n_steps)]
        mu_voltages = [0]*n_steps
        
        for i in range(n_steps):
            mu_voltages[i] = self.zotino0.voltage_to_mu(voltage, self.offset_dacs,self.vref)
        
        self.set_zotino_voltages(self.zotino0, mu_voltages)    

    def run(self):
        self.core.reset()
        self.test_zotinos()

```

I expected this to output a ramp wave on the zotino, going from 0 to 10V in 0.1V steps. When I manually convert the voltages to machine units, it runs as expected. 
When I try to run this it gives the following error.
AttributeError: 'Zotino' object has no attribute 'voltage_to_mu' 

Having looked in the source code, I don't think the `voltage_to_mu()` is imported into the Zotino source file from the AD9910 file. Could this be why it isn't working?

I'm running ARTIQ v5.6852.c2b4f0cf.beta on Windows 10

jordens, 2019-07-17:
`voltage_to_mu` is not a method of the `Zotino` class.

See https://m-labs.hk/artiq/manual-beta/core_drivers_reference.html#artiq.coredevice.ad53xx.voltage_to_mu

Use:

```python
from artiq.coredevice.ad53xx import voltage_to_mu
```

No AD9910 here.

cnourshargh, 2019-07-17:
What's the logic behind not having voltage_to_mu as a method of the zotino?

jordens, 2019-07-17:
That would restrict it to only that usage. Might be ok though.

sbourdeauducq, 2019-08-26:
What about making it a method of ``AD53xx``, like the DDS ``frequency_to_ftw``?
Importing directly from ARTIQ like this breaks the device database abstraction.

PR #1445 (2020-05-08): ad53xx improvements

# ARTIQ Pull Request

## Description of Changes
*  Input validation has been added to voltage_to_mu(). (closes #1444)
*  Documentation improvements
    * State valid voltage range (closes #1443)
    * Conversion is only valid for 16-bit members of the family
*  Make `voltage_to_mu()` availible as a `statimethod`. The exising function is maintained for backwads compatibility, but is also exposed as a `staticmethod` of `AD53xx`. (closes #1341)

## Type of Changes

|   | Type |
| ------------- | ------------- |
|  | :bug: Bug fix  |
| ✓  | :sparkles: New feature |
|   | :hamm [... cut]
commit b3b6cb8e (2020-05-08): ad53xx improvements (#1445)

* ad53xx: voltage_to_mu() validation & documentation (closes #1443, #1444)
The voltage input (float) is checked for validity. If we need more
speed, we may want to check the DAC-code for over/underflow instead.

* ad53xx documentation: voltage_to_mu is only valid for 16-bit DACs

* AD53xx: add voltage_to_mu method (closes #1341)

* ad53xx: improve voltage_to_mu performance
Interger comparison is faster than floating point math.

* AD53xx: voltage_to_mu method now uses attribute values

* Fixup RELEASE_NOTES.rst

* ad53xx: documentation improvements

voltage_to_mu r [... cut]
closed 2020-05-08 by jordens