m-labs/artiq #1343: Integrate SU-Servo in kasli_generic EEM descriptions

jordens, 2019-07-17:
# ARTIQ Feature Request

## Problem this request addresses

All supported EEMs (except Novogorny) apart are representable in the JSON based system configuration in kasli_generic. SU-Servo needs a custom fully elaborated Python variant definition and can't be mixed with the JSON-style definitions.

## Describe the solution you'd like

Support using Sampler and Urukul-AD9910 with SU-Servo in `kasli_generic`.

closed 2019-09-11 by sbourdeauducq

hartytp, 2019-09-11:
> https://freenode.irclog.whitequark.org/m-labs/2019-09-09#25320433

@cjbe/@dnadlinger should correct me if I'm wrong, but I believe the answer is "yes, we do that here without issues". The only observation I have is that the coredevice driver would need a tweak to continue supporting this if we moved to non-blind CPLD initialization in a future revision.

dnadlinger, 2019-09-11:
@hartytp: I only use two Urukuls per SUServo – in fact, we are currently working on supporting up to four at once. I suspect it should work, though, since there is no acknowledgement on any of the buses.