m-labs/artiq #1344: ad9910: expose/abstract DRG (ramp generator)

jordens, 2019-07-24:
# ARTIQ Feature Request

## Problem this request addresses

The AD9910 DDS on Urukul has a digital ramp generator that is not exposed in the coredevice API.

## Describe the solution you'd like

Abstract/Expose it.

## Additional context

Similar level of abstraction/exposure as the RAM methods.
Details TBD.

spomjaksilp, 2020-10-09:
I will start working on this feature in the upcoming days.
Are the DRG control pins exposed to `artiq` in any way? The Urukul schematics imply that they are at least connected in rev 1.5.

jordens, 2020-10-09:
You will probably need to change the Urukul CPLD gateware. Making it backwards compatible would be key.

spomjaksilp, 2020-10-09:
thx for the hint. Did not think about that :+1: