m-labs/artiq #1350: Config for Kasli v1.1 vs v1.0

TheCakeIsAPi, 2019-08-14:
# Question

What changes, if any, need to be made to the configuration when replacing a Kasli v1.0 with Kasli v1.1?

## Description

Our config is patterned after Opticlock.  In kasli.py we have the line:

`        if hw_rev is None:
            hw_rev = "v1.0"`

But it is not obvious where hw_rev will otherwise be assigned.  Is it meant to be passed in as a command line argument?

Then, this flag doesn't seem to be used for anything besides:

`        if hw_rev == "v1.0":
            # EEM clock fan-out from Si5324, not MMCX
            self.comb += self.platform.request("clk_sel").eq(1)`

Is there a different condition that should be specified for v1.1.  In particular we are using an external clock.

TheCakeIsAPi, 2019-08-14:
I think more generally it looks like I should probably be using the "JSON system description file" with kasli_generic.py, but I'm at a loss without any documentation or examples for that.

sbourdeauducq, 2019-08-15:
There are examples here:
https://git.m-labs.hk/m-labs/sinara-systems
See also https://github.com/m-labs/artiq/issues/1346#issuecomment-518314607.

closed 2019-08-15 by sbourdeauducq

TheCakeIsAPi, 2019-08-15:
Thank you, I was able to switch over to using kasli_generic with those examples!  Perhaps include that link in the documentation?