m-labs/artiq #1352: Silent failure for Urukul rf switch with too many simultaneous events

charlesbaynham, 2019-08-16:
# Bug Report
## One-Line Summary

Changing the state of the RF switch fails silently if there are already too many events at the same time in the RTIO queue. 

## Issue Details

Presumably due to finite number of RTIO lanes. Code to reproduce:

```
from artiq.experiment import *
from artiq.coredevice.core import Core
from artiq.coredevice.ad9912 import AD9912

class DDSFailMWE(EnvExperiment):

    core = None  # type: Core
    
    def build(self):
        self.setattr_device("core")

        self.setattr_argument("num_ttls", NumberValue(default=7, step=1, ndecimals=0))

        self.setattr_device("urukul0_ch1")

    def prepare(self):
        self.ttls = [self.get_device("ttl{}".format(i)) for i in range(4, 4 + self.num_ttls)]

    @kernel()
    def run(self):

        ttls = self.ttls
        dds = self.urukul0_ch1  # type: AD9912

        freq = 10.0e6

        self.core.reset()
        dds.init()
        dds.set_att(0.)
        self.core.break_realtime()

        ##############

        # This sequence will work fine

        dds.set(freq)
        dds.cfg_sw(True)
        delay(10 * ms)
        dds.cfg_sw(False)

        ##############
        delay (30 * ms)
        ##############

        # This sequence will silently fail to activate the RF switch if num_ttls > 6

        dds.set(freq)

        for t in ttls:
            t.on()

        dds.cfg_sw(True)
        delay(10 * ms)
        dds.cfg_sw(False)

        ##############
        for t in ttls:
            t.off()
        delay(30 * ms)
        ##############

        # This sequence will be ok, even if num_ttls > 6

        dds.set(freq)

        for t in ttls:
            t.on()

        delay_mu(10)

        dds.cfg_sw(True)
        delay(10 * ms)
        dds.cfg_sw(False)

        for t in ttls:
            t.off()

```

### Expected Behavior

Either 3x 10 MHz RF pulses of 10 ms each, or an error message explaining why sequence two must fail. 

### Actual (undesired) Behavior

In sequence two, the RF switch is not opened, but no error is raised:

[image]

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: `ARTIQ v5.6910.5f8acb3f.beta`
* Version of the gateware and runtime loaded in the core device: ` 5.0.dev+567.g99e490f9;npl2`
* If using Conda, output of `conda list` : [conda_output.txt](https://github.com/m-labs/artiq/files/3501160/conda_output.txt)
* Hardware involved: Kasli + 2x Urukul, 2x SMA-DIO, 2xSampler, 1x Zotino

dnadlinger, 2019-08-16:
Are you getting a sequence error in the core log? If so, this is (unfortunately) expected.

charlesbaynham, 2019-08-16:
Nope, completely silent, hence my concern. Here's a full debug log:

[dump.log](https://github.com/m-labs/artiq/files/3510393/dump.log)

dnadlinger, 2019-08-16:
Just to confirm, the core device log is completely silent? It wouldn't show up on the dashboard unless you have set up aqctl_corelog; can check directly on Kasli UART to verify. Asynchronous exceptions, such as RTIO sequence errors, are only reported there.

charlesbaynham, 2019-08-16:
Ah interesting: I rebooted the rack to check that my UART connection was working properly and now I am getting the expected RTIO sequence errors appearing both in the UART and in the dashboard. I also restarted the artiq_master (and client) but I don't think that's the cause, since I've been restarting those all day long. That's the first power cycle that I've done of the hardware today though.

charlesbaynham, 2019-08-19:
So I suppose this issue is actually about a failure to receive corelog messages, both on UART and in the dashboard. BTW, I do seem to get the corelog errors showing up in the dasboard now that I've restarted the system. 
I'd rename this issue to something more relevant, but I'm know knowledgable enough about ARTIQ / Sinara to choose an appropriate name. Would someone else like to?

jordens, 2019-08-19:
Can you describe how to reproduce this?

charlesbaynham, 2019-08-19:
I'm afraid I probably can't: I've yet to reproduce it. It happened after having the system running for ~ a day, uploading a variety of experiments, etc. I'll post back here if I manage to get it to happen again.

charlesbaynham, 2020-07-05:
I've never seen this issue again so I'll close it.

closed 2020-07-05 by charlesbaynham