m-labs/artiq #1354: Documentation for basic use of each of the Boards

cnourshargh, 2019-08-19:
# ARTIQ Feature Request

## Problem this request addresses

The documentation for using the EEM boards is at a very high level and is doesn't clearly show what is required to use each board. This has made it very difficult top start using the language. I have heard that there are other people who have encountered the same problems as me and it seems that very few people who are experienced in using ARTIQ are being relied upon to show many different people individually how to do some relatively basic things, because there is no documentation for this. 

## Describe the solution you'd like

I think it would help if there were a page for each of the EEMs and for Kasli, outlining what the board is capable of doing and how to start using it, including some example code. I am happy to add some of the details to these if a template can be provided. 

## Additional context

I have already been working on minimum working examples of code for each of the boards in my Sinara device, along with some documentation to help people who haven't used ARTIQ before, and it seems to make sense to try and collaborate on these.

charlesbaynham, 2019-08-19:
I agree that the learning curve is quite steep at the start (I'm still climbing it!). I'd be happy to contribute to these too.

jordens, 2019-08-19:
@rnourshargh hast started collecting material for novice users. Maybe you can collaborate.

dtcallcock, 2019-08-19:
We've just been through the pain of getting an Urukul and Sampler working and this is sorely needed.   I saw [this repository](https://github.com/cnourshargh/Bham-ARTIQ-examples) which looks like a great start.  A majority of the issues we had though were related to the device_db, different hardware versions, and hardware settings (like bitstreams, clocking options & DIP switches).  I think that information needs incorporating somehow too.

One of our students @aquinn2 would very much like to help.  In particular we'd like to get some example SUServo code up and @hartytp is helping us with this over [on the forum](https://forum.m-labs.hk/d/19-suservo-sample-code/2).

What is envisaged as the final location for this?  Should it be integrated into the documentation as @cnourshargh suggests or can it just exist as a repository as it does now?

hartytp, 2019-08-19:
I think that he most useful thing to cover the basics is a new Sinara section in the manual with a page for each board/su-servo etc

That should contain a link to the pdf schematic, he wiki page, very brief description of current state of artiq support (maybe a link to the relevant git hub label for open issues) etc then it should contain a tutorial style section that explains basic bring up, followed by more advanced uses (eg urukul phase sync) along with descriptions of what needs to be done to configure hardaware (jumpers clocks etc) and device dB. 

I think for these basic uses a tutorial is much more valuable than code snippets

RNourshargh, 2019-08-20:
That all sounds really good @hartytp and @dtcallcock . If a template or starter page in the right place can be made in the next day or so then we will be able to make a contribution by the end of the week. After that @cnourshargh will be leaving birmingham and we will loose some momentum. Is a template, or even a page with a title something you could add @jordens?

hartytp, 2019-08-20:
Also worth debugging is things like (using urukul as an example) where to find cpld code and it’s docs (as well as a brief overview of what it does/why it’s there) and what diagnostic info one has (red led means roughly that dds is configured, green means rf switch on etc)

aquinn2, 2019-08-22:
Could something along [these](https://github.com/aquinn2/artiq/blob/master/doc/manual/getting_started_sinara.rst) lines be of interest?

I'm imagining a document with an introduction going over some of the basics of interfacing EEM with a host and with one another.  This is followed by a section for for each Sinara board discussing both 1) how set the the board up and 2) how to control this board using ARTIQ.  I've (mostly) filled out a section for SMA/BNC DIO boards to give a sense of what I'm thinking.

Do y'all think a format like this might work?  I'm not too sure what the best way to organize things would be, but it seems like since there's a lack of clear information at both the hardware and software levels, it would make sense to discuss hardware and software issues separately.  Maybe subsections discussing common issues/debugging would be good to add as well?

gkasprow, 2019-08-22:
One could have a look at how manuals of other equipment are organized, for example from National Instruments.
You can also place [front panel drawings](https://www.dropbox.com/sh/ibpkmvff5kuahs2/AACwiob3X5PPDderlNJB3qdia?dl=0) so users can quickly identify the module in the rack visually.
DIP switches positions are critical for successful operation, so I'd recommend placing photos or drawings with the default configuration. For example, DIO modules direction can be controlled by I2C only when DIP switches are off.
One should mention something about the power supply (there is a wiki about it), Ethernet connection (link to the wiki with SFP transceivers), clock routing (when it is necessary).
I'd make a wiki page with sub-pages dedicated to particular modules, or even write it in Latex so later could be printed and kept on hand. 
Maybe we should consider an extension of current wiki product pages by template sections?

dtcallcock, 2019-08-22:
I think one of the underlying reasons for the documentation gap is that it's unclear (to me at least) where the Sinara wiki documentation should end and where the Artiq manual should begin.

jordens, 2019-08-26:
Good tutorials/examples/board documentation should:

* Be testable or self-testing. I.e. something like doctests or high quality tooling that is run in CI. Otherwise the examples diverge from the code and/or users are battling version mismatches.
* The documentation should also live in close proximity to the code/the hardware designs. That eases joint editing and synchronization between docs and code/designs.
* In my opinion good documentation will need to be viewed as a comprehensive package that should cover and reference datasheet-style properties, description of the devices, operating modes, manual-style instruction sections, all the way to the examples.
* Should be done in such a way that typeset manuals can be generated. I'm fine with the ARTIQ manual being the one-stop-shop for documentation.

If we have a development/maintenance/template style, I'd be happy to look at adding it.
Maybe the first step is to discuss what the content of the documentation should be.

jordens, 2020-01-21:
A couple more example repositories:

* https://github.com/chase1635321/ARTIQ-suservo
* https://github.com/chase1635321/ARTIQ
* https://github.com/HaeffnerLab/artiq-work-lattice
* https://github.com/vontell/nvOS
* https://github.com/GuanQunMu/IonTrap-WIPM