m-labs/artiq #1363: Failed to set the frequency above half of the clock frequency for AD9914 after upgrading from 3.6 to 3.7

hpy0226, 2019-09-19:
## Failed to set DDS(AD9914) frequency above half of the clock frequency

We used to be able to directly set DDS frequency with any value below the clock frequency ~ 2.7 GHz on artiq 3.6. 

But after upgrading from version 3.6 to 3.7, we failed to set the DDS frequency above 1.35 GHz. If we tried to set frequency above that, the DDS frequency would be set to 1.35 GHz, which looks like an upper limit. 

We could still pre-convert the frequency with MHz to a number with machine unit, and then we can use set_mu function to set DDS frequency with any frequency we want, although the artiq GUI panel would show a negative frequency value which is the Nyquist image signal with a minus sign. For example, if the DDS is set to be 1.5 GHz, you would see the frequency is -1.2GHz =-(2.7-1.5)GHz on the GUI. 

I suspect the function 'frequency_to_ftw' on artiq 3.7 doesn't work in the same way on artiq3.6 which might limit the frequency range? or are there other changes for 3.7 related to this issue?

sbourdeauducq, 2019-09-23:
2c12e150f387383820d6c67177d8009f9345d930 maybe? @whitequark 
You can print the output of ``frequency_to_ftw`` on the two versions and try to narrow down the problem. 
Note that, generally, the ARTIQ DDS support has been designed for use in the first Nyquist zone, so there may be other problems.

hpy0226, 2019-09-25:
okay, we don't have a second machine running on artiq 3.6. Thanks for reminding the potential phase mode issues.