m-labs/artiq #1364: Error on import with sub-experiment and pandas

charlesbaynham, 2019-09-19:
# Bug Report

## One-Line Summary

In a specific set of circumstances dependent on the order of Experiment scanning, either a TypeError or a RecursionError is produced when artiq scans the experiment repository. 

## Issue Details

### Steps to Reproduce

1. Load the experiment repository at https://github.com/charlesbaynham/pandas-import-mwe
2. (conda package environment is specified in `environment.yml` if required)
3. Start a new artiq_session using artiq version 4e77be05. 
4. Observe error when experiment directory is scanned. 

### Expected Behavior

No error

### Actual (undesired) Behavior

Trace of error: 

[image]

Full trace of error:
[err.log](https://github.com/m-labs/artiq/files/3631490/err.log)

### Odd fix

Now, rename one of the experiment files (e.g. `z.py`) to `a.py`. It is now imported before the sub-experiment `imported_exp.py`, and the error does not occur:
[image]

### Requirements

The circumstances to produce this error seem to be:

* Use of a sub-experiment by >1 Experiment stored in separate modules
* Import of the `typing` module in each Experiment module
* Import of pandas and creation of a DataFrame in build() of the sub-Experiment
* Naming such that the sub-Experiment is parsed before either of the top-level E.experiments

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: 4e77be05
* Version of the gateware and runtime loaded in the core device: n/a - no core device required
* If using Conda, output of `conda list`: see `environment.yml` in linked repo. 
* Hardware involved: None

sbourdeauducq, 2019-09-23:
Thanks for the detailed report.

This seems to be a problem specific to the ``typing`` module of Python 3.5. The problem is gone with the new ``typing`` module in 3.7 that does not have this ``__extrahook__`` method anymore (maybe for this reason?)

We will most likely upgrade the conda packaging for Python 3.7 (https://github.com/m-labs/artiq/issues/652) as part of a NIST contract, but in the meantime are you able to use ARTIQ-5 + Linux + Nix, which already supports 3.7?

charlesbaynham, 2019-09-23:
I'm probably stuck with windows for now, but I can work around it by just not importing `typing` and removing any type hints from my functions. 

For anyone else who finds this issue, I used to use
```python
from typing import
if TYPE_CHECKING:
    from mod import Stuff
```

But, on pycharm at least, 

```python
# noinspection PyUnreachableCode
if False:
    from mod import Stuff
```

also results in useful autocompletion. 

Type hints for function parameters can be moved to the docstring with 
```
:param x: A number
:type x: int
```

sbourdeauducq, 2020-12-02:
@charlesbaynham Is this also fixed with the new Py3.7 packages?

charlesbaynham, 2021-05-08:
Yes, this is fixed!

closed 2021-05-08 by charlesbaynham