m-labs/artiq #1370: Compiler performance: Seconds spent creating/deleting global variables; cache get_abi_{size, alignment}

dnadlinger, 2019-10-08:
On our experiment, compilation times have recently reached upwards of 15 seconds, which is a serious impediment to getting any work done interactively, and even starts to be limiting for automated calibrations.

Looking through the compiler output, the experiments don't actually hit any pathological case in terms of code produced. On inspecting the profile, 20% of the time (of the overall `artiq_run` total) is spent in llvmlite's `_get_ll_pointer_type` function, which is called from `get_abi_{size, alignment}`, which are used during general codegen (`dereferenceable` metadata) and attribute writeback synthesis.

This is because `_get_ll_pointer_type` is implemented in an exceedingly inefficient way: To obtain the LLVM type needed to query `TargetData`, it emits a (textual) LLVM module with a global of the type in question, and then queries the type of that. This is clearly not a good idea for a function anywhere near the hot path. (I suspect the reason might have been a reluctance to extend the llvmlite Python/C shim library with the one or two functions necessary to directly construct a type.)

Either way, this is all just to say that llvmlite's `Type.get_abi_{size, alignment}()` are slow, and we should memoize/cache their results.

(py-spy generated flamegraph and speedscope profiles attached in case anyone is curious – the repro case is a bit harder to share: [artiq_run.speedscope.zip](https://github.com/m-labs/artiq/files/3703340/artiq_run.speedscope.zip) [artiq_run.svg.zip](https://github.com/m-labs/artiq/files/3703341/artiq_run.svg.zip))

whitequark, 2019-10-08:
Yes. llvmlite is quite bad. I was not happy with it back when we decided to use it (the string concatenation it uses is also fairly pathological in nature and shows up a lot in profiles), but IIRC @sbourdeauducq insisted that it should be done that way to save time.

sbourdeauducq, 2019-10-09:
More like to get anything to work at all. We have limited resources especially for the compiler (the expectations of the compiler have been increasing since the very first NIST contracts but subsequent external funding for it is essentially zero, and compiler devs are hard to find). We can't reinvent every component right from the start, and I replacing llvmlite with something better later should not involve rewriting much of the compiler (since it is just a binding for the LLVM API anyway). llvmlite was already an improvement over llvmpy back then...
If you want to rewrite llvmlite now, please go ahead, just do not break anything in ARTIQ master until everything is ready.

dtcallcock, 2019-10-09:
> replacing llvmlite with something better

Is there a reasonable alternative?

> rewrite llvmlite

How much work would it entail to get a significant improvement here?

> We have limited resources especially for the compiler (the expectations of the compiler have been increasing since the very first NIST contracts but subsequent external funding for it is essentially zero

Should we be thinking of a way for the community to fund this (and other less exciting areas of Artiq that have been neglected recently)?  Hopefully all those groups who recently bought Sinara crates are an opportunity.

whitequark, 2019-10-09:
> replacing llvmlite with something better later should not involve rewriting much of the compiler (since it is just a binding for the LLVM API anyway)

That's not true. llvmlite is *not* a binding for the LLVM API, which is the whole problem here. It is a completely new API for emitting textual LLVM IR. Replacing llvmlite would involve changing about half of the LLVM emitter in the compiler.

dnadlinger, 2019-10-09:
What's the latest in Python LLVM bindings anyway – is there one that supports the (C++) IRBuilder interface?

whitequark, 2019-10-09:
> What's the latest in Python LLVM bindings anyway – is there one that supports the (C++) IRBuilder interface?

I wouldn't try to bind to the C++ interface, since it's unstable and changes a lot. The whole point of the C interface is that it is reasonably stable between LLVM versions--after all I'm the maintainer of it. There's been some attempt to autogenerate Python bindings to the C interface, it is not published as a package on GitHub or anything, but I have it in my mail somewhere. It could be useful as a starting point.

dhslichter, 2019-10-09:
Very roughly, how much effort (in hours) would you estimate to replace the existing llvmlite with something better at a production-ready performance level?  Compiler pain is definitely a major issue for every ARTIQ user I have spoken to, and it seems that we should be able to drum up some funding for work on this from among all the different groups that are currently using ARTIQ.  I understand that there is also a bandwidth/personnel issue to be considered, as @whitequark has his plate full already.

dnadlinger, 2019-10-11:
> I wouldn't try to bind to the C++ interface, since it's unstable and changes a lot. 

What I meant is something _like_ the C++ IRBuilder interface, which would not necessarily consume the C++ interface itself under the hood. (Instantiating everything manually is a pain, and `llvmlite` already offers a similar interface.)

As an aside, while it's true that the C++ interface changes over time, the required client code changes really aren't too bad most of the time. Even supporting multiple versions concurrently is manageable (we have been doing it in LDC for many years now), although that does tend to require a few `#ifdef`s here and there. We only target a single LLVM version anyway, though…

sbourdeauducq, 2019-10-11:
> This doesn't affect the issue at hand but shows the cost of forking.

Not forking also has a cost; in most cases, upstream developers aren't going to maintain our code for free.

sbourdeauducq, 2019-10-11:
> The generic solution is probably either some way of (implicitly) caching intermediate/partial compilation results or (explicitly) building libraries of independently compiled user code. All this doesn't sound so easy because it clashes with the isolation between experiments (workers/runs and repo versions).

What about a cache that consists of a global (for the whole master) hash table "textual source code of some module/class/function + variable values -> binary that can be linked"? That would not really clash.

dnadlinger, 2019-10-11:
>  But 20% of compile time is still not that much. 

Yes – but a _very_ low-hanging fruit (patch incoming), given that what the functions actually need to do is really just reading a few fields from DataLayout (née TargetData).

> It's not clear to me that going for the LLVM C bindings or using the C++ IRBuilder brings scalability of the concept. What am I missing?

You aren't missing anything; two issues are being conflated in this thread. In terms of long-term sustainability/funding goals, I would definitely suggest looking into architectural features such as the ones you mentioned. In the meantime, however, if we can get a factor of a few speedup by fixing the most egregious issues (together with surveying our experiment code for needless volumes of code), this will push compilation times into livable territory for our current stage of complexity.

dhslichter, 2019-10-11:
> What about a cache that consists of a global (for the whole master) hash table "textual source code of some module/class/function + variable values -> binary that can be linked"? That would not really clash.

This would be useful, although it only covers a limited number of cases because we oftentimes will change just one or two variable values which don't have any impact on the timing (e.g. we change the values of DDS frequencies which are set during a kernel, while leaving all timing alone) -- this would produce a different hash even though it shouldn't need to go through the whole compilation process again.  It might be helpful to have flags for variables where the user can specify that the timing is unaffected by these variables, and you guarantee a specific type for those variables, and then cache a compiled kernel which just gets the values of the slotted variables put in, without having to do the whole compilation again?  

I think in general there are a lot of compiler optimizations that would be good to implement, and it will inevitably entail some kinds of flags or hand optimizations or things which may be a bit fragile.  I am OK with having some optional features like this making compilation more fragile, with the idea that expert users can trade some fragility for increased performance if they desire.  

Anyway, these are things that I think we are all aware of.  Would it be a good idea to start a meta-issue on compiler optimizations that would be good to do?  There are a lot of individual issues, and they are flagged as area:compiler for easy finding, but perhaps a more general discussion about the types of optimizations, design choices, tradeoffs, etc might be useful?  I'm curious as to your thoughts @sbourdeauducq @whitequark @jordens @dnadlinger

closed 2019-10-28 by dnadlinger

jordens, 2019-10-28:
Great. Thanks!

commit 611bcc4d (2019-10-28): compiler: Cache expensive target data layout queries

On one typical experiment, this shaves about 3 seconds (~25%)
off the overall kernel compilation time.

GitHub: Closes #1370.