m-labs/artiq #1372: Scheduling experiment accessing urukul channel causes ConnectionResetError

schmidf, 2019-10-11:
# Bug Report

## One-Line Summary

Scheduling an experiment that uses an Urukul AD9910 channel causes a `ConnectionResetRerror` in the preceding experiment.

## Issue Details

### Steps to Reproduce

1. Submit the following experiment twice using the Dashboard:

```python
from artiq.experiment import *

class BreakingExample(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul0_ch0")

    @kernel
    def run(self):
        self.core.reset()
        delay(5*s)
        self.core.wait_until_mu(now_mu())
```

### Expected Behavior

The experiment runs twice, each time doing nothing for 5 seconds.

### Actual (undesired) Behavior

The first scheduled experiment fails with a ConnectionResetError. Only the second one runs completely. Error messages for the first experiment:
```
root:Terminating with exception (ConnectionResetError: Connection closed)
Traceback (most recent call last):
  File "C:\Users\Be\AppData\Local\conda\conda\envs\artiq-5\lib\site-packages\artiq\master\worker_impl.py", line 288, in main
    exp_inst.run()
  File "C:\Users\Be\AppData\Local\conda\conda\envs\artiq-5\lib\site-packages\artiq\language\core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Users\Be\AppData\Local\conda\conda\envs\artiq-5\lib\site-packages\artiq\coredevice\core.py", line 137, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "C:\Users\Be\AppData\Local\conda\conda\envs\artiq-5\lib\site-packages\artiq\coredevice\comm_kernel.py", line 501, in serve
    self._read_header()
  File "C:\Users\Be\AppData\Local\conda\conda\envs\artiq-5\lib\site-packages\artiq\coredevice\comm_kernel.py", line 116, in _read_header
    (sync_byte, ) = struct.unpack("B", self.read(1))
  File "C:\Users\Be\AppData\Local\conda\conda\envs\artiq-5\lib\site-packages\artiq\coredevice\comm_kernel.py", line 99, in read
    raise ConnectionResetError("Connection closed")
ConnectionResetError: Connection closed

artiq.master.scheduler:got worker exception in run stage, deleting RID 804
```

### Your System (omit irrelevant parts)

* Operating System: Windows 7
* ARTIQ version: v5.6991.ebd5d890.beta
* Version of the gateware and runtime loaded in the core device: 5.0.dev+599.g0b1fb255;mpq
* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long)
[conda_list.txt](https://github.com/m-labs/artiq/files/3717283/conda_list.txt)

* Hardware involved: Kasli v1.1, Urukul v1.3 (AD9910)

dnadlinger, 2019-10-11:
Is there anything on the core log (easiest to verify on the UART interface using the front-panel USB connection)?

sbourdeauducq, 2019-10-11:
> easiest to verify on the UART interface using the front-panel USB connection

You don't even need that, just run ``artiq_coremgmt -D device_ip log``

dnadlinger, 2019-10-11:
True – in this case, the comm CPU seems to be (mostly) happy, so the network interface should work.

sbourdeauducq, 2019-10-11:
Also this experiment does not actually access the Urukul channel, and it could be random behavior from this bug: https://github.com/m-labs/artiq/issues/1125

schmidf, 2019-10-11:
Here is the core log with log_level DEBUG: [core_log_connection_issue.txt](https://github.com/m-labs/artiq/files/3718173/core_log_connection_issue.txt). I believe that the first experiment was transmitted at [    46.172805s], and the second one at [    47.285039s]. This seems to result in the first connection failing with a timeout error a second later:
```
[    48.299852s] DEBUG(smoltcp::socket::tcp): #1:192.168.0.100:1381:192.168.0.1:50647: timeout exceeded
```

I then repeated the process with the line `self.setattr_device("urukul0_ch0")` in the experiment commented out. The two experiments are transmitted at [    59.674745s] and at [    64.881655s] and no error occurs.

I also tried calling setattr_device for various devices and could only reproduce the error with urukul[x]_ch[y]. For instance, instantiating urukul0_cpld, led0, and ttl0 work fine.

sbourdeauducq, 2019-10-11:
> I then repeated the process with the line `self.setattr_device("urukul0_ch0")` in the experiment commented out. The two experiments are transmitted at [ 59.674745s] and at [ 64.881655s] and no error occurs.

Does it do that every time or is this just one data point?

schmidf, 2019-10-11:
This behaviour is completely reproducible. I always get a `ConnectionResetRerror` on the currently running experiment when scheduling another experiment that calls `setattr_device("urukul[x]_ch[y]")` in `build()`. I haven't seen any timeouts otherwise.

sbourdeauducq, 2019-10-11:
Ah, you have Urukul sync, right?
That could be the EEPROM read that is causing the issue.

schmidf, 2019-10-11:
I belive so. We have bought the AD9910 version and the DDS outputs are synchronized to the RTIO timestamps.

A hardware access during build would explain why scheduling an experiment can disturb the currently running one.

sbourdeauducq, 2019-10-11:
Can you disable synchronization (as a workaround and to confirm that this is indeed the problem) until we fix this? Remove ``sync_delay_seed`` and ``io_update_delay`` from device_db.py.

schmidf, 2019-10-11:
Indeed the issue does not appear with synchronization disabled.

dnadlinger, 2019-10-11:
As a workaround (i.e. if you need synchronisation to work right now), you could also directly specify the two values in the device db.

schmidf, 2019-10-11:
That sounds like a good workaround. How can I read out the EEPROM values and in which format do they have to be specified in the device_db?

dnadlinger, 2019-10-11:
See the `Urukul` constructor for how to access the EEPROM, but you could also just read them from the member variables after the fact. Alternatively, you can just re-run the `tune_*` functions to find the parameters, like in the following (hacky) script:

```py
class AutotuneDelays(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ccb")

        self.name = self.get_argument("Device", StringValue("urukul0"))

    def prepare(self):
        self.cpld = self.get_device(self.name + "_cpld")
        self.channels = [
            self.get_device(self.name + "_ch{}".format(i)) for i in range(4)
        ]

    @kernel
    def run(self):
        self.core.reset()
        delay(30 * ms)
        self.cpld.init()

        for i in range(len(self.channels)):
            ch = self.channels[i]
            print()
            print("Channel", i)
            print("---")

            self.core.break_realtime()
            print("msync (delay, window):", ch.tune_sync_delay(15))

            self.core.break_realtime()
            print("io_update_delay:", ch.tune_io_update_delay())
```

commit 5279bc27 (2019-11-05): urukul: rework EEPROM synchronization. Closes #1372
closed 2019-11-05 by sbourdeauducq