m-labs/artiq #1373: Fastino RTIO gateware support

jordens, 2019-10-12:
# ARTIQ Feature

## Problem this request addresses

Support [Fastino](https://github.com/sinara-hw/Fastino) as an ARTIO RTIO device.

## Describe the solution you'd like

* Kasli gateware: serializer, bit and word clocking concept, DAC update generation
* Kasli RTIO Phy: RTIO channel number and address partitioning, concurrent streaming of DAC updates on all channels at constant rate, concurrent setting of all DAC channels, support for 32 DAC channels per Fastino board, support for 2 MS/s DAC data rate per channel for all channels concurrently, enable/disable of data streaming and/or DAC clocking, RTIO event rate limited by existing RTIO infrastructure. ARTIQ EEM definition.
* ARTIQ coredevice driver: To support data and clocking disable/enable, conversion between machine units and SI units, data packing, RTIO event generation and submission.

## Additional context

WIP prototype work:
* [Fastino gateware](https://github.com/quartiq/fastino)
* [ARTIQ RTIO Phy](https://github.com/m-labs/artiq/tree/fastino)

Funded by Oxford.

jordens, 2020-01-23:
Some random test code for early adopters:

```python

from artiq.experiment import *
import numpy as np

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("fastino0")

    @rpc(flags={"async"})
    def p(self, *p):
        print([hex(_) for _ in p])

    def run(self):
        n = 1 << 2
        self.s = [self.fastino0.voltage_to_mu(9.9*np.cos(2*np.pi*i/n))
            for i in range(n)]
        self.do()

    @kernel
    def do(self):
        self.core.reset()
        f = self.fastino0

        self.core.break_realtime()
        f.init()

        delay(1*us)
        f.set_hold(0 << 27)
        delay(10*us)

        for ch in range(32):
            f.set_dac_mu(ch, 0x8000) # ch << 11)
            delay(1*us)

        delay(100*us)
        f.set_dac(27, 0.) # ch << 11)
        return
        f.update(0xffffffff)
        delay(1*us)

        for i in range(10):
            f.set_leds(0xaa)
            delay(.1*s)
            f.set_leds(0x55)
            delay(.1*s)

        k0 = 14*7//2*8*2
        ch = 27
        m = 1 << ch

        for i in range(1000000):
            for j in range(len(self.s)):
                f.set_dac_mu(ch, self.s[j])
                delay_mu(k0)

        for i in range(10000):
            for v in range(1 << 16):
                f.set_dac_mu(ch, v)
                delay_mu(k0)

        self.core.wait_until_mu(now_mu())

```

sbourdeauducq, 2020-10-15:
Is this finished?

jordens, 2020-10-15:
There is a refactor of the link in #1533 but this is done.

closed 2020-10-15 by jordens