m-labs/artiq #1374: Better interpolation for DACs

jordens, 2019-10-12:
#  ARTIQ Feature Request

## Problem this request addresses

* Spline interpolation is used in PDQ and SAWG.
* Interpolation is required to allow a compact representation of waveform data.
* It's not very resource efficient as a spline knot is typically around 10 times the size of a single sample.
* Computation of spline knots is expensive
* The details of the mode of interpolation are typically not critical.
* There is a need for dynamically changing the data rate (before interpolation) and supporting burst input data as there are long phases of no action between intervals of high data rate.
* Usual FIR or IIR interpolation filters can't be used due to their more-or-less fixed characteristics.

## Describe the solution you'd like

I'd like to consider using variable-rate CIC interpolators for amplitude/frequency on SAWG and amplitude in Fastino.

* CIC interpolators can perform PDQ- or SAWG-style spline interpolation
* They do it pretty much for free in real-time without the need to supply any additional coefficients.
* They can do higher order filtering compared e.g. to the Milldown linear ramps or PDQ cubic splines.
* They have no over/undershoot (c.f. FIR/IIR filters).
* They have a well-known transfer function that is fully amenable to analysis and pre-emphasis (DC electrode decoupling network compensation).
* Could be combined with a configurable real-time FIR pre-emphasis filter. But that's a different feature.
* They are extremely resource efficient.
* You can also use them to deliberately and dynamically place transfer function zeros on/near secular or other frequencies of interest.
* They don't have the radical stopband attenuation or flat passband that you can get from generic FIR/IIR filters but that doesn't seem to matter here.
* We'd need to give the time domain multiplexing on Fastino some thought but that seems doable.
* We'd build a CIC filter that supports variable upsampling rate (powers of two or arbitrary with a gain compensation multiplier)
* The CIC filter would support burst input data transparently.
* Sharing of and updates to upsampling rate among channels TBD depending on application.

## Additional context

The usual DSP literature on CICs.

jordens, 2019-12-10:
funded by LUH

hartytp, 2019-12-10:
Cool!

dhslichter, 2019-12-10:
Very nice!  When appropriate, can you please post a more detailed spec of what you will implement @jordens?  One feature that would be extremely useful on our end (indeed, for anyone doing ion transport) would be to have the ability to add on an FIR pre-emphasis filter, and so we may consider an additional contract to implement this feature.  Would this be straightforward to add on after the fact with the design you are currently envisioning?

jordens, 2019-12-10:
As mentioned above, that's a different feature. This will be just a configurable CIC for Fastino. You'll have to see whether what you want to do is easy or difficult. I have no idea yet.
The assumption that everybody doing transport needs to do preemphasis is questionable. It relies on a weird idea that it's good to cancel the hardware lowpass with a software highpass. As mentioned above, you will want to design that into the waveform since also the CIC is not flat and it's frequency characteristic changes depending on the configuration which may change during the waveform.

dhslichter, 2019-12-10:
> The assumption that everybody doing transport needs to do preemphasis is questionable. It relies on a weird idea that it's good to cancel the hardware lowpass with a software highpass. 

Not sure why this is "questionable" or "weird" -- it seems pretty straightforward if one starts with the notion that trap electrodes are almost always aggressively low-pass filtered (c.f. basically all ion traps in use today), and that transport on timescales faster than the characteristic time of the filter response is useful for some applications (more debatable, but pursued by some groups).  Anyway, designing it into the waveform rather than doing it in at the DAC interpolation stage is fine, I was just curious about the possibilities as it seems like a potential opportunity to reduce the need for centralized pre-compute, especially if what you are doing is making small tweaks to voltage waveforms based on e.g. changing stray fields, which might not be so large as to necessitate re-optimization of an entire waveform scheme.

sbourdeauducq, 2019-12-11:
Will this supersede the current ``artiq.wavesynth`` code?

jordens, 2019-12-11:
> Will this supersede the current artiq.wavesynth code?

No. This only targets fastino for now.

jordens, 2019-12-11:
Demanding both aggressive lowpass filtering and aggressive highpass filtering to compensate for the former is questionable because they counteract. Since you fully control the lowpass and its design, instead of asking how to undo your own filter choice ex post, you may want to ask yourself how to make the right filter and how to implement it. Undoing your own filter choice costs dynamic range, bandwidth, distortion, computation time, noise, and gateware complexity. Evaluating filter topologies or peaking networks is priceless.

Not sure why you need to re-calculate the pre-emphasis when doing small tweaks if you don't even need to re-optimize for small tweaks.

dhslichter, 2019-12-11:
> Since you fully control the lowpass and its design, instead of asking how to undo your own filter choice ex post, you may want to ask yourself how to make the right filter and how to implement it. 

Absolutely true unless you already have filters in vacuum (put there by someone decades ago, perhaps) and can't/don't want to break vacuum.  Also, there is trepidation from some corners about things like notch filters (rather than just aggressive low pass) because it's new/different.  I'm not saying that predistortion for aggressive lowpass is the right thing until the end of time (I think you're right that it's not); it is just something that would probably be useful right here and now with traps we are running.  I would contend that there will probably always be some low-pass (e.g due to cryostat wiring resistance, or frequency-dependent amplifier gain, for example), where predistortion might be desirable even if it's not "aggressive".  

> Not sure why you need to re-calculate the pre-emphasis when doing small tweaks if you don't even need to re-optimize for small tweaks.

For example, you have a desired transport waveform to appear on the electrodes.  The optimization process takes into account max voltages and slew rates, with some margin of safety.  If you want to slightly shim the desired transport waveform, you can just predistort the shimmed waveform, rather than redoing the entire optimization from scratch with the shimmed waveform as target (if you are confident that you won't exceed the margin of safety w/r/t voltage/slew rate by doing so).  Then you could do scans of transport over shim strengths, or transport time, for example, without having to pre-compute different waveforms for each scan point on a PC and pass all those data down.  

Anyway, I am not trying to argue with you that there aren't better ways to do transport and filtering than are currently used!  I am trying to contend that given the systems we are working with today, predistortion that is directly integrated with the digital waveform synthesis process could have some useful applications.  I understand that there are limitations and subtleties, e.g. the non-flat and configuration-dependent behavior of the CIC, and those might just tank it.

jordens, 2019-12-12:
Sure. You need a Rf decoupling capacitor close to the trap. But the external filter should account for that and flatten the transfer function for the frequencies of interest (below Rf). If you **also** use that RF decoupling capacitor to protect the ion from your noisy DAC at the secular frequencies  then you obviously can't do fast things.

If the shim is static you don't need to preemphasize it. If it is dynamic you don't need to recompute anything because you will already have added preemphasize to the shim waveforms and linearity does the rest.

dhslichter, 2019-12-13:
> If it is dynamic you don't need to recompute anything because you will already have added preemphasis to the shim waveforms and linearity does the rest.

Depending where your transport starts and ends, you may be moving through a certain region of the trap at different speeds, and also if you are changing speeds (e.g. sweeping transport duration) then the required preemphasis will be changing as well.  In both such cases it's not quite as simple as the above.  The value of having the pre-emphasis calculated on the fly at the DAC as the last step is that it would cover the above use cases more efficiently than having precalculated variants of the waveforms and shims for all the different cases above.

jordens, 2019-12-13:
If you are transporting fast and you want to change speed, you will likely need to re-optimize your waveform to keep control of the motional excitation. It's not just a change in timescale.

And it's certainly not what I would call "small tweaks to voltage waveforms based on e.g. changing stray fields".

dhslichter, 2019-12-14:
I'm talking about a time scan through a small range (as a fraction of total transport duration) to find optimum transport duration, and small adjustments to shims (~10 V/m) to be as compensated as possible throughout the duration of a transport.

jordens, 2019-12-15:
That's not what I would do. A static shim is just static in time independent of its spatial shape and whether it's local or global. It's application does not need any spectral shaping independent of ist strength. A dynamic shim has the same challenges as the master waveform. And addition of waveforms is always strictly linear. They would be shaped independently.
Changing waveform speed is much more complicated if you don't want aliasing and don't want to be restricted to integer interpolation ratios. There is no simple knob to do it. It needs a resampler.